mne-tools/mne-python #10708: Enable blank GH issue templates

hoechenberger, 2022-06-03:
I just wanted to open an issue that wouldn't fit into any of our existing categories:

[image]

So I was looking for something like I've seen on other repositories: a link that allows you to create a blank issue without a template, e.g. like this on conda-forge:

[image]

Can we please enable this?

agramfort, 2022-06-03:
I don't know how to do this

> Message ID: ***@***.***>
>

hoechenberger, 2022-06-03:
No need to, @drammock already figured it out!

PR #10710 (2022-06-03): allow blank issues
closes #10708

cc @hoechenberger
closed 2022-06-03 by hoechenberger