mne-tools/mne-python #10709: Releasing MNE-Python 1.1

hoechenberger, 2022-06-03:
I'd love to see MNE-Python 1.1 released within the next 1–2 weeks, unless there are any hard blockers. Could all of you please take a look at open issues and PRs and add those that you think should definitely be addressed for 1.1 to the [respective milestone](https://github.com/mne-tools/mne-python/milestone/22)?

I've also created a 1.2 milestone to allow us to just move other stuff we'd want to keep on our roadmap, but that's not critical for 1.1.

ping @agramfort @larsoner @adam2392 @alexrockhill @sappelhoff @cbrnr @drammock @britta-wstnr

hoechenberger, 2022-06-03:
To create the installer bundles, we also need to ensure that MNE-BIDS etc are readily compatible with 1.1

If their latest stable release is not compatible, we'll need to cut new releases there too

drammock, 2022-06-03:
mailmap update at #10711

hoechenberger, 2022-06-03:
CC @marsipu, forgot to add you above

larsoner, 2022-06-04:
I looked through the issues and didn't see anything that we need to consider a blocker actually, so I re-milestoned everything for 1.2. So everyone feel free to swap the order of operations now: rather than bump things to 1.2 that aren't blockers, look at the 1.2 list and change back to 1.1 anything that you think we should really get in within the next week or so, ideally with a comment about who should/will work on it.

alexrockhill, 2022-06-04:
It would be nice to fix the blender tutorial links at least here #10684

mmagnuski, 2022-06-05:
Since we've added `mne.viz.plot_ch_adjacency` I think I could add a short example about its interactive usage.

hoechenberger, 2022-06-05:
@mmagnuski Care to make it a VIDEO?? I wanna make a bunch of small clips soon, having one on adjacency could be nice too!!! Just a minute or two ...

mmagnuski, 2022-06-06:
@hoechenberger I originally thought about text + gif, but I can try :) Should it be with or without voice? Do we have any examples of how it should look like?

larsoner, 2022-06-06:
@alexrockhill could you take a look at fixing/editing the blender tutorial?

alexrockhill, 2022-06-06:
Yes I will as soon as I get a chance but it will probably be tomorrow.

hoechenberger, 2022-06-07:
> @hoechenberger I originally thought about text + gif, but I can try :) Should it be with or without voice? Do we have any examples of how it should look like?

A short video (why GIF? even Twitter supports "proper" videos these days – making it easier to interact with, e.g. to rewind a bit etc) is a great idea! I like the approach of using a of screen recording (that's what you had in mind, right?) + added text!

From the Twitter website:

> Maximum file size is 512MB for TweetVideo, but you are able to upload a video longer than 2 minutes and 20 seconds, and trim it before including the video in a Tweet.

mmagnuski, 2022-06-07:
@hoechenberger 
Thanks, I was asking because I would like it to be part of our docs, not only a video that can be posted somwhere. So I was thinking about the example to have all the relevant code and text cells and only the interactive aspect would be shown in the video (this would likely make the video < 30 s). Do you mean the text should be added to the video stream? Do you have any tools to suggest where adding such text is easy?

hoechenberger, 2022-06-13:
Only #10284 and #10742 are left to be resolved, and I believe the installers are also ready!

cbrnr, 2022-07-27:
I added #10963, this is ready and should be green very soon!

PR #10994 (2022-08-03): prep for release 1.1
closes #10709
closed 2022-08-03 by drammock