mne-tools/mne-python #10713: Getting ready for Python 3.11

hoechenberger, 2022-06-03:
I'm not sure how much we'll actually benefit from this as most of our time-consuming code lives in NumPy and other highly-optimized libraries, but the upcoming performance improvements in CPython are quite impressive IMHO:

https://docs.python.org/3.11/whatsnew/3.11.html#faster-cpython

I think it would be worthwhile to keep this in mind and maybe start testing with a pre-release of Python 3.11 soon to ensure we'll be compatible – and, of course, to assess whether and to what degree we will benefit from those performance improvements :)

larsoner, 2022-06-04:
We can look to https://github.com/numpy/numpydoc/pull/375 for how to do this, but it doesn't look all that complicated. Ideally we could switch over the pip-pre job first once there are SciPy and NumPy prerelease wheels there

larsoner, 2022-06-06:
No NumPy prerelease wheels in https://pypi.anaconda.org/scipy-wheels-nightly/simple/numpy/, let's wait until they show up at least

hoechenberger, 2022-10-27:
Python 3.11 ha been released. 

Changes:
https://docs.python.org/3/whatsnew/3.11.html

Python 3.7 will be EOL in 8 months:
https://endoflife.date/python

hoechenberger, 2022-10-27:
The better trackbacks will be super helpful esp for beginners, I think we should try to include Python 3.11 in our installers as soon as we can.

larsoner, 2022-10-27:
I'd expect a few month lag for all dependencies to have releases that are 3.11 compatible. But we can start trying now and restart CIs periodically to see where there's still a problem. I'll get the ball rolling in mne-installers

hoechenberger, 2022-10-27:
Sounds great!

hoechenberger, 2022-10-27:
FYI the conda-forge feedstock transition is also still WIP:

https://conda-forge.org/status/#python311

hoechenberger, 2022-10-27:
Oh and conda-forge is dropping Python 3.7:

https://conda-forge.org/docs/user/announcements.html#dropping-python-3-7

larsoner, 2023-07-31:
We now test 3.11, closing!

closed 2023-07-31 by larsoner