mne-tools/mne-python #10724: Tutorial thumbnail popovers should be removed

hoechenberger, 2022-06-08:
The popover texts seem to be based on the raw ("un-processed") input, so for the devel docs I get e.g.:

[image]

And for the stable docs the CSS seems to be broken:

[image]

Besides, for those tutorials where I don't see such layouting issues, the provided information is not helpful to me at all as it's just too little text. If we want more info, we should amend the captions of the thumbnails.

Besides, I don't like stuff jumping all around on my screen as I move the mouse cursor over it anyway, it's irritating; it hides content; and the added motion is, in fact, a big problem for folks out there with motion sickness. The theme doesn't respect my macOS settings for reduced motion, so as long as this isn't fixed anyway, we should not have this jumpy stuff :)

So my vote is clear – let's get rid of those popovers entirely.

@drammock WDYT?

drammock, 2022-06-08:
The content for those is drawn from the first sentence of the tutorial text. I agree it is of limited use. At some point @larsoner had done some magic in sphinx-gallery to make it "sanitize" those preview snippets so that backticks and tildes would not render, but it looks like it's not working in all cases anymore.  I have no objection to suppressing them. Can you see if there's a config option for sphinx-gallery that will just turn them off?

larsoner, 2022-06-08:
We can just set display none in the CSS for the class. It's easier than adding yet another option to SG

hoechenberger, 2022-06-08:
Sounds good to me!!

PR #10727 (2022-06-09): MRG: Remove tutorial thumbnail popovers
Fixes #10724

cc @larsoner @drammock
commit 255cc03e (2022-06-09): Remove tutorial thumbnail popovers (#10727)

Fixes #10724
closed 2022-06-09 by larsoner