mne-tools/mne-python #10725: Specify vmin vmax color range in animated topomap

cdussard, 2022-06-09:
#### Describe the new feature or enhancement
Add the possibility to specify the value bounds of the colormap range in the _mne.Evoked.animate_topomap_ function.
The _mne.viz.plot_topomap_ function implements vmin and vmax for static topomaps but there seems to be no equivalent in the animated version of the function.

#### Describe your proposed implementation
 _mne.viz.plot_topomap_  function implements vmin and vmax parameters (lines 1700-1704 from topomap.py)
```
vmin = np.min(vlims)
    vmax = np.max(vlims)
    cmap = _setup_cmap(cmap, n_axes=n_times, norm=vmin >= 0)
```

animate_topomap could implement vmin and vmax parameters in the same way. They could be passed on from to mne.Evoked.animate_topomap to the params dictionnary (line 2361-2369 of topomap.py)
```
params = dict(data=data, pos=pos, all_times=evoked.times, frame=0,
                  frames=frames, butterfly=butterfly, blit=blit,
                  pause=False, times=times, time_unit=time_unit,
                  clip_origin=clip_origin,vmin=vmin,vmax=vmax)
[...]
 animate_func = partial(_animate, ax=ax, ax_line=ax_line, params=params)
```
as the __animate_ function already seems to allow vmin and vmax into params : (lines 2267-2268 of __animate_ function
    vmin = params['vmin']
    vmax = params['vmax']

Sorry if this isn't clear, this is my first PR 👀 

#### Additional comments
*Add any other context or screenshots about the feature request here.*

agramfort, 2022-06-09:
Sounds very legitimate. Would you consider sending a PR?

Thx

hoechenberger, 2022-06-09:
If you don't have time, we can also try to address this during this year's code sprint :)

cdussard, 2022-06-09:
I can definitely try to do a PR.
I can find some time but I may lack the skills on how to handle the PR
properly, I'll look it up!

drammock, 2022-06-09:
> is this in the right direction ? https://mne.tools/stable/install/contributing.html

yes!  As you're working on this, please also take note of anything missing from that page that you needed to know in order to make your first PR. That will help us improve the contributing guide for future first-time contributors!

drammock, 2022-08-22:
@matsvanes is working on this

PR #11073 (2022-08-24): Add `vmin` and `vmax` specification to `mne.Evoked.animate_topomap`
Fixes #10725 .

Add `vmin` and `vmax` specification to `mne.Evoked.animate_topomap`
closed 2022-08-24 by drammock