mne-tools/mne-python #10743: Add FieldTrip channel "neighbors" for standard 10-05, 10-10, and 10-20 montages, and beef up read_ch_adjacency() visibility

hoechenberger, 2022-06-12:
`mne.channels.read_ch_adjacency()` reads FieldTrip neighbors stored in `mne/channels/data/neighbors`. It appears that currently, we're only shipping a subset of the [neighboring definitions](https://www.fieldtriptoolbox.org/template/neighbours/) that FT provides. Importantly, we currently don't include their 10-05, 10-10, and 10-20 neighbor definition, which I think would be of great use for EEG users.

Specifically, I propose we include the following files (quoted from the FT website linked to above)

> The standard 10% system
> 
> Standard 10-05 system (elec1005_neighb.mat)
> Standard 10-10 system (elec1010_neighb.mat)
> Standard 10-20 system (elec1020_neighb.mat)

I have to admit that I was never really aware of the fact that we even **did** ship adjacency matrices like these. I always assumed `read_ch_adjacency()` was only for reading back adjacency matrices I had previously created manually or with the help of `find_ch_adjacency()`. Now it turns out there's an easier way, and one could potentially skip using @mmagnuski's (excellent!) interactive manual adjacency adjustment tool and simply use one of FT's matrices we already include, at least in numerous situations! 

I also propose to add a function similar to `get_builtin_montages()` to get the names of built-in FT adjacency matrices – `get_builtin_adjacencies()`, perhaps?

Happy to split up this issue into multiple tasks / sub-issues, depending on your feedback.

cc @sappelhoff @mmagnuski @agramfort @larsoner

sappelhoff, 2022-06-12:
+1 to add the 10-05 etc. adjacencies.

Furthermore, I don't know if FieldTrip uses "CODEOWNER" files, but if they do, we should add some of us to subscribe to this directory: https://github.com/fieldtrip/fieldtrip/tree/master/template/neighbours ... just so that we get notified in case there is a change that we should mirror to our copies of the adjacencies.

> I also propose to add a function similar to get_builtin_montages() to get the names of built-in FT adjacency matrices – get_builtin_adjacencies(), perhaps?

would be fine with me -- but I'd also be fine with removing `get_builtin_montages()`. I think it's enough to document the included montages / adjacencies in `make_standard_montage` and `read_adjacency` respectively

> Lastly, read_ch_adjacency() is not really demo'd anywhere, so I guess it's no wonder I never really had an idea … we should include it in a tutorial!

👍 I tried to link a bit more to it in #10658 via "See Also", but yes: making it more visible in a tutorial would be another good step

hoechenberger, 2022-06-13:
> would be fine with me -- but I'd also be fine with removing `get_builtin_montages()`. I think it's enough to document the included montages / adjacencies in `make_standard_montage` and [...]

Oh noes, I **just** killed that docstring you're referring to and reworked the way we store that information … wait, you won't understand what I'm talking about hehe, so here's the PR with those changes: #1074

Do you feel this is not the right way to go?

mmagnuski, 2022-06-13:
@hoechenberger I think you meant #10737 :)
I saw that PR when it got merged, but I was not sure removing the descriptions from the docstring was a good idea. I think it is often more convenient to look up the docs of the function you are using (ctrl + I in spyder) than type in another function and browse through its outputs. The docstring also rendered nicely as a table.
But other than this I agree. :)

hoechenberger, 2022-06-13:
> @hoechenberger I think you meant #10737 :)

Oh, yes, sorry! I've edited my post to correct for that!

sappelhoff, 2022-06-13:
Ah I see :thinking: I am with @mmagnuski on this one: I like it better in the docstrs. But it's not a strong preference, I can also live with a consistent set of functions that print out the available options :shrug:

hoechenberger, 2022-06-13:
> I saw that PR when it got merged, but I was not sure removing the descriptions from the docstring was a good idea. I think it is often more convenient to look up the docs of the function you are [...]

I can try to reconstruct that table automatically in the docdict via the `get_builtin_montages()` function, WDYT? (never done sth like this before, but it might work!)

sappelhoff, 2022-06-13:
> I can try to reconstruct that table automatically in the docdict via the get_builtin_montages() function, WDYT? (never done sth like this before, but it might work!)

both docstr and printing functions without unneeded duplication sounds good to me :)

hoechenberger, 2022-06-13:
> Ah I see 🤔 I am with @mmagnuski on this one: I like it better in the docstrs. But it's not a strong preference, I can also live with a consistent set of functions that print out the available options 🤷

Ok, that's already two!

I will see what I can do about this.

hoechenberger, 2022-06-13:
@sappelhoff I have to say I'd also feel quite uncomfortable to be listed as a "code owner" in a repo I've never contributed to :)

I think a trivial way to resolve this is by adding a CI job that maybe runs once a week. It boils down to:

- checking out MNE-Python main
- checking out FieldTrip master
- diff'ing `fieldtrip/template/neighbors` with `mne/channels/data/neighbors`
- failing the job if they don't match anymore, and sending out an email

sappelhoff, 2022-06-13:
Yep, I think that's a good idea!

I think `CODEOWNER` is just a bad misnomer for a file that's intended to make notifications easy if particular files are touched in a repo :roll_eyes: I totally get your point

PR #10749 (2022-06-16): MRG: Sync FieldTrip channel adjacency matrices with FT repository, and improve "read_ch_adjacency" discoverability
Closes #10743

WIP:
- [x] improve `read_ch_adjacency()` discoverability
- [ ] update docstrings
- [x] add `get_builtin_adjacencies()`?

cc @sappelhoff
closed 2022-06-16 by hoechenberger