mne-tools/mne-python #10747: Retrieving channel adjacencies for our standard montages

hoechenberger, 2022-06-12:
I'd like to be able to do something like the following:

```python
# %%
import mne

montage_name = 'easycap-M1'

m = mne.channels.make_standard_montage(montage_name)
adj = m.get_adjacency()
```
i.e., if I'm using one of our built-in montages and we are shipping a matching adjacency matrix, I want to be able to retrieve this matrix without having to worry about `ch_names` etc in the way I have to when using the current approach:

```
# %%
import mne

m = mne.channels.make_standard_montage('easycap-M1')
adj, ch_names = m.get_adjacency('easycapM1')
```

Alternatively, we could have `make_standard_montage(..., return_adjacency=False)` return the matrix alongside the montage.

WDYT?

cc @mmagnuski @sappelhoff @cbrnr

sappelhoff, 2022-06-12:
I like your first suggestion -- that seems like a reasonable method to have!

hoechenberger, 2022-06-12:
> I like your first suggestion -- that seems like a reasonable method to have!

But it implies that the DigMontage "knows" it was created from one of our standard montages (and which). There is currently no way to store that information, or is there?

sappelhoff, 2022-06-12:
> But it implies that the DigMontage "knows" it was created from one of our standard montages (and which). There is currently no way to store that information, or is there?

it could simply look for an exact match of channels between montage and "adjacency", right? 🤔

hoechenberger, 2022-06-12:
> it could simply look for an exact match of channels between montage and "adjacency", right? 🤔 

I'm not so sure, as there might be different adjacency matrices for the exact same list of channel names! At least in theory... no?

Also we'd have to load all adjacency matrices one by one until we find a match, this seems quite inefficient?

sappelhoff, 2022-06-13:
How about a private attribute in the montage class then?

hoechenberger, 2022-06-13:
I am notdürftigere it's so easy, as we store montages as FIFF files and you know how it is with this standard :/

Or could we maybe transition to something else? HDF5 or so? @larsoner

mmagnuski, 2022-06-13:
I'm not sure that makes things so much easier. If you have montage name, you can just do:
```python
montage_name = 'easycap-M1'

m = mne.channels.make_standard_montage(montage_name)
adj = mne.channels.read_ch_adjacency(montage_name)
```
assuming `read_ch_adjacency` can do it. If it can't it would be nice if it could. :)

mmagnuski, 2022-06-13:
(ah, ok I get your point with `ch_names`)

hoechenberger, 2022-06-13:
> assuming `read_ch_adjacency` can do it. If it can't it would be nice if it could. :)

also don't forget about #10746 💩