mne-tools/mne-python #10750: DOC: new tutorial for using automated preprocessing pipelines / switching back and forth between EEGLAB and MNE-Python

dominikwelke, 2022-06-13:
#### Proposed documentation ehancement
hi All,

for the 2022 sprint I suggested to add a small tutorial on using MNE-Python with data that has been preprocessed in other software (i.e. interoperability).
I particularly thought of EEGLAB which knows a bunch of neat fully-automatic preprocessing pipelines for EEG data.

It's not very difficult to import these data, and all the tools are there. yet I thought that a small dedicated tutorial in the I/O section might highlight this fact for new users, thus making it even easier to 1) try MNE coming from EEGLAB or 2) use the strenghts of different softwares for different subtasks.

there are several options how to implement such a tutorial, from a minimum version with a few lines of code for importing preprocessed EEGLAB data, to a longer and more general version with best practices on interoperability, or different other softwares respectively.

@drammock suggested to open this issue in advance to discuss it a little: 
what do you think, any comments, suggestions? 

p.s. by a quick scan I found several potentially related issues in the board (EEGLAB only): 
#9989 #10505 #10482 #8581 #6610 #3837

larsoner, 2023-10-01:
@dominikwelke I can't remember if we talked about this at the sprint or not, but I agree this would be useful indeed if you or someone else is interested in looking into it!

withmywoessner, 2024-02-08:
@larsoner @dominikwelke I am very interested in this.