mne-tools/mne-python #10767: Changed color scheme?

hoechenberger, 2022-06-15:
Hello, I'm not sure if this was introduced upstream in our theme or we did that ourselves, but I don't really like the new colors here:

[image]

Aside from the reduced contrast to compared what we had before for headlines, now links and headlines have the same color, so I caught myself actually clicking on headlines a few times, believing that's some sort of hyperlink. I don't think this is a great choice.

drammock, 2022-06-15:
It was an upstream change and I didn't want to executively change it, and I also didn't want to open that can of worms just before the 1.1 release 😆

hoechenberger, 2022-06-15:
So am I correct in assuming you're not too fond of this change either?

drammock, 2022-06-15:
> So am I correct in assuming you're not too fond of this change either?

I'd like to change a lot of our colors, actually. The syntax highlighting colors of our code blocks changed sometime between stable and current dev docs, but it happened when I was away on leave and I haven't gone digging yet to figure out what caused it. I liked the way it was on the 1.0 docs.  But yeah, I agree the new primary color is a bit low-contrast on light backgrounds (though looks good on dark) and I can see the point about different colors for headings and links.  Also those colors are pydata "brand" colors so there's an argument that we probably *should* overwrite them.

hoechenberger, 2022-06-15:
> The syntax highlighting colors of our code blocks changed sometime between stable and current dev docs, but it happened when I was away on leave and I haven't gone digging yet to figure out what caused it. I liked the way it was on the 1.0 docs

Same! The new layout looks quite "dated" to me!

Is the color scheme something we could try to attack during the sprint?

drammock, 2022-06-15:
> Is the color scheme something we could try to attack during the sprint?

I think it's something you and I should work out, and then in theory anyone could implement it (but it would be easier for one of us to do it).  So possibly, depends on how much time is needed supporting the participants. Maybe I'll tag it for the sprint and assign it to you, to keep it on our radars.

PR #10874 (2022-07-27): website theme updates
WIP / do not merge. Working on consistency with upstream theme enhancements.

closes #10767
closed 2022-07-27 by drammock