mne-tools/mne-python #10775: Complete FieldTrip neighbors descriptions

hoechenberger, 2022-06-15:
We've added lots of new FieldTrip neighbors (adjacency) definitions and corresponding descriptions in #10749

However, for a number of the neighbors, I was unable to find a proper description, and the respective field remained blank (empty string). We should aim to get them filled.

Maybe this could be part of our upcoming sprint? @drammock I'll leave it to you to judge :)

drammock, 2022-06-15:
where would folks look up the description if it's not obvious to them what it ought to be?

hoechenberger, 2022-06-16:
I took the existing descriptions from the FT website, maybe I overlooked some. Hence, my approach would be the following (in that order):

- For each built-in FT neighbors file, check if it exists in the [FT repository](https://github.com/fieldtrip/fieldtrip/tree/master/template/neighbours)
- If it does, check if there's any description on https://www.fieldtriptoolbox.org/template/neighbours/ that I maybe simply overlooked
- If there isn't one, probably ask on the FT issue tracker (?) as to how those "unknown" neighbor definitions were created
- If the file does **not** exist on the FT repository, ask the MNE developers if they know where that file comes from; potentially use the Git history feature to figure out who added the file to MNE, and when