mne-tools/mne-python #10780: ENH: Add offliine homogeneous field correction for OPMs

larsoner, 2022-06-17:
Should be simple enough to implement, and should make data for OPMs cleaner, at least for whole-head arrays / multi-axis measurements

https://www.sciencedirect.com/science/article/pii/S1053811921007576

larsoner, 2022-10-24:
Looks like I already had this issue open @neurofractal @georgeoneill , so let's discuss here.

> This is great news to hear [about field correction]! As you can see we've been doing some initial tests. I've been trying to work out whats the best approach for MNE-Python. We normally do a hard correction to the data (rather than having an online montage) and then update our 'compensation matrix' to update our forward modelling later. But I realise this is not in the nature of how the SSP filters work in MNE-Python (i.e. only activated when needed). This might need some thought.

Yes we'll have to be a bit careful. I have not looked at the math at all but doubt they are true orthogonal projection vectors, so wouldn't be good to call SSP; and I doubt they work like compensation matrices for CTF (which have some nice mathematical property -- like its inverse is its transpose or something like this, I'd have to check to remind myself). We can probably add a new constant for it -- we've wanted one before for the cross-talk correction for MEG data anyway, which is itself just a matrix multiply (that does not modify/reduce rank).

So I say in a second PR after #11257 is done let's get it doing what it needs to do -- probably as a matrix that can be applied (or not) just like a projection -- and then I can talk to some FIF folks about the right way to store the information.

I think the answer to both of the following is "yes", but just to check -- can the higher-order version also be described as just a matrix multiply? And they are not time-varying (i.e., just depend on sensor geometry, not sensor position as a function of time), right?

georgeoneill, 2022-12-21:
Hello @larsoner, starting to think about how to best to tackle this, now we have an importer working. 

One _potential_ issue we have going forward (other than requiring new classes to represent arbitraty sensor balancing) is the licensing of the original code which generated the basis sets. A simple relicensing and port of `spm_opm_vslm.m` (which is what SPM uses to generate the regular vector spherical harmonics) is a no-go unfortunately. 

However, there's probably a couple of BSD-compatible solutions. 

1) Build upon MaxFilter module basis set (specifically the _outer_ space). 
2) Use the [bfieldtools](https://bfieldtools.github.io/) module, which has functions specifically for geneating vector spherical harmonics. I _think_ this approach means we don't need to do any cartesian to spherical coordinate transforms too. 
3) Simply just implement the original Homogenous Field approach where the basis set are the sensor orientations, lose access to gradient, quadratic terms. 

Which of these sounds the most appealing?

georgeoneill, 2022-12-21:
I thought this might be the answer! I'll do some exploration.

PR #11536 (2023-04-22): ENH: Add support for Harmonic Field correction
Fixes #10780.

#### What does this implement/fix?
- Add support for [Homogenous Field Correction](https://www.sciencedirect.com/science/article/pii/S1053811921007576) and [extending the harmonic basis set to higher orders](https://www.sciencedirect.com/science/article/pii/S1053811922004578). 
~~- Creates an `FieldCorrector()` class which can be extend to correct forward models (important for deeper sources).~~
- Generates a `Projector()` for each column of the basis set.

#### To do
1) Add unit test(s)
2) update OPM preprocessing tutorial to incor [... cut]
closed 2023-04-22 by larsoner