mne-tools/mne-python #10782: When reading EGI data (.mff), import channel names from sensorLayout.xml

hoechenberger, 2022-06-18:
When working on https://mne.discourse.group/t/what-eeg-montage-to-use-for-257-channels/5093/11, I found that we currently don't use the channel names provided in (all? some?) MFF datasets. Specifically, `read_raw_egi()` always applies a custom naming scheme, ignoring all available metadata. This led, for example, to the following channel, `REF`, being named `E257` on import.

Snippet from `sensorLayout.xml`:
```
        <sensor>
            <name>VREF</name>
            <number>257</number>
            <type>1</type>
            <x>0.000</x>
            <y>-0.000</y>
            <z>0.000</z>
            <identifier>1001</identifier>
        </sensor>
```

scott-huberty, 2022-06-24:
Hi @hoechenberger, thanks for bringing this up, I saw that discourse thread, it's definitely an issue that has come up for me, particularly when setting montages. This might overlap with the issue I raised [10823](https://github.com/mne-tools/mne-python/issues/10823).

It's not just MNE-Python, EEGLAB also names these channels according to their number (at least in my experience). The only issue I foresee, is that even if we change the code in `read_raw_egi()` to import  the channel names from `sensorLayout.xml`, users will still hit a snag when setting a standard montage, because in the MNE-Python standard Montages for EGI nets, the `REF` channel is named `Cz`. 

Not sure what the best approach is here. Maybe we do amend the code to import the channel names from `sensorLayout.xml` as you suggest, and then add `VREF` to `mne.io.constants.CHANNEL_LOC_ALIASES`, so that users have the option skip the step of manually renaming the `VREF` channel before setting a standard montage with `raw.set_montage`. But at the same time, I'm not sure how many users will actually be aware of this nifty `match_alias` parameter in `raw.set_montage`, which defaults to `False` : ) 

Or maybe we just make sure that the channel names in the standard montage for EGI nets match the channel names that are imported during `read_raw_egi`(regardless of whether we decide to keep it as `E257`, or change it to `VREF` or even `Cz`). Not sure! But i'm happy to help with whatever you all think is the best approach during the code sprint next week.

mscheltienne, 2022-06-25:
I'll drop this code snippet that could be useful:

```
import xml.etree.ElementTree as ET
from pathlib import Path

import numpy as np
from mne.channels import make_dig_montage

# load montage
# fname is the path to a .mff folder
montage_fname = Path(fname) / "coordinates.xml"
# the montage contains 257 channels with the field <number> from 1 to  257
# and 3 fiducials: Nasion, LPA, RPA with the number and names:
# - Nasion - 258
# - Left periauricular point - 259
# - Right periauricular point - 260
namespaces = {"": "http://www.egi.com/coordinates_mff"}
tree = ET.parse(montage_fname)
root = tree.getroot()
sensors = root.findall(".//sensorLayout/sensors/sensor", namespaces)
ch_pos = dict()
for sensor in sensors:
    name = sensor.find("name", namespaces).text
    number = sensor.find("number", namespaces).text
    name = f"E{number}" if name is None else name
    name = "E257" if name == "VREF" else name
    x = sensor.find("x", namespaces).text
    y = sensor.find("y", namespaces).text
    z = sensor.find("z", namespaces).text
    ch_pos[name] = np.array([float(x), float(y), float(z)])
nasion = ch_pos["Nasion"]
lpa = ch_pos["Left periauricular point"]
rpa = ch_pos["Right periauricular point"]
del ch_pos["Nasion"]
del ch_pos["Left periauricular point"]
del ch_pos["Right periauricular point"]
montage = make_dig_montage(ch_pos, nasion, lpa, rpa, coord_frame="unknown")
```

Note that the loaded montage is not correctly scaled.

If EGI saves the channels as `f'E{k}'` + `'VREF'`, I think MNE should load those names instead of renaming the reference channel to e.g. `E257`.

scott-huberty, 2022-06-28:
Thanks for sharing the code! I've only done some preliminary testing on my local machine, but I think I have amended `mne.io.read_raw_egi()` to load the chan names where the ref channel is now called `VREF`

As far as the channel locations go, it's seems like there is already code in `egimff.py` that [reads the sensor locations](https://github.com/mne-tools/mne-python/blob/1310269f27bdf4347dfc465a834c1e04d25563d8/mne/io/egi/egimff.py#L257-L295).

I'm first trying to wrap my head around how `read_raw_egi()` currently uses these sensor locations. It seems like at some point in post processing, you usually need to set a standard montage, using the one provided by mne-python for EGI nets. should the channel locations in `coordinates.xml` be used for the montage? And if that's the case, do we need to explicitly set the montage inside `read_raw_egi`?

It might be helpful to chat with one of you some time this week!

scott-huberty, 2022-06-29:
Apologies, I was mistaken - you don't need to set a standard montage.

 See below, plotting the sensors, with the amended code to read in the channel names, with the ref chan named `VREF`:
[image]

Compared to the prior code, with the ref chan named `E129`:
[image]

and finally, plotting the sensors after setting a standard montage `GSN-HydroCel-129`:
[image]

PR #10898 (2022-07-21): use channel names in read_raw_egi and set the reference location
closes #10869 (supersedes)
closes #10782