mne-tools/mne-python #10784: trans parameter in make_field_map() should accept a Transform and path-like

hoechenberger, 2022-06-19:
Currently only accepts a str, which can be a path to `'auto'`, whose behavior is not documented.

Path should not be a str either, but `path-like`, so this needs fixing too.

hoechenberger, 2022-06-19:
Turns out the docstring is wrong … is DOESN'T work with a path, it expects a `Transform`. So this is a bug, I suppose…

hoechenberger, 2022-06-19:
Argh ok it's more complicated than that. The docstring is outdated.

- It does work with a str containing a path
- It **does not** work with a `Path`
- If does work with a `Transform`

larsoner, 2022-06-19:
To me this is a pretty simple fix to use `_get_trans` like we do elsewhere. It should automatically support everything you want. And then the docstring should be deduplicated / use `%(trans)s`. The function just came way before this helper and hasn't been updated

PR #11071 (2022-08-23): Fix trans loading logic in make_field_map
Hello - this is my first PR - let me know if I've missed anything

#### Reference issue
Example: Fixes #10784.

#### What does this implement/fix?
This PR changes the loading of the trans information in `make_field_maps` and `plot_alignments`. This now uses the standard `mne.transforms._get_trans` logic and preserves the possibility to use `trans='auto'` to load from a specified fs subject directory. 

Docstrings have been updated and the `test_field_map_ctf` now loads the trans via a `pathlib.Path`.

Where these functions tested for `trans is None`, w [... cut]
closed 2022-08-23 by drammock