mne-tools/mne-python #10790: Realistic montage for EGI systems

mscheltienne, 2022-06-20:
@hoechenberger Following the discussion on the forum [here](https://mne.discourse.group/t/what-eeg-montage-to-use-for-257-channels/5093/12), maybe we could add this realistic montage for EGI systems: https://sites.google.com/site/cartoolcommunity/files/EGI257.GenevaAverage13.10-10.xyz?attredirects=0&d=1

It was measured several years ago (2011? 2013?) at HUG in Geneva by Laurent Spinelli, and is an average between 13 subjects wearing a Hydrocel 257 (256 electrodes + ref) cap in an MRI. I'm trying to get more information.. but it's not super recent.

The format/scaling will need some work if you want to include it.

hoechenberger, 2022-06-20:
@cbrnr WDYT? I'm not sure if we'd want this or maybe do our own coreg on fsaverage instead and then use **those** locations?

cbrnr, 2022-06-20:
I would guess that if someone doesn't have individual MRIs, the better choice would be fsaverage, because that's much more widely used. It would be great if we had a 10–05 montage coregistered to fsaverage (and we could do other systems like EGI)!

mscheltienne, 2022-06-20:
`fsaverage` sounds like a way better choice!

closed 2022-06-20 by mscheltienne