mne-tools/mne-python #10791: MNE IO RAW Decimate

umair-hassan, 2022-06-20:
Dear MNE Maintainers,

I am trying to decimate the raw data (loaded using raw.io), I do not care about the antialiasing and filters. I just need the decimated date (downsampled ratio basically). I am not sure if I can create epoch easily for the whole data and then downsample it, therefore I think inside resample (there could be an option to switch off the filters) or there could be a generic decimate method for raw io object as well.

Happy to help implementing it though. 

Best, Umair

agramfort, 2022-06-21:
@umair-hassan look at the decim parameter in mne.Epochs class

close the issue if it fixes your issue

umair-hassan, 2022-06-21:
I need it for a concurrent EEG fMRI artefact correction step. I cannot just apply antialiasing in the presence of huge gradient artefacts. The data was sampled at exceptionally high Fs therefore downsampling it to a reasonable Fs is also necessary. I will take care of anti-aliasing myself.

I think generally speaking for all neuromodulation, TMS-EEG analysis too, this would be quite desirable.

cbrnr, 2022-06-21:
You can always grab the continuous data with `raw.get_data()` and then decimate the NumPy array yourself. Would that work?

umair-hassan, 2022-06-21:
Yes in principle.

larsoner, 2022-06-21:
People do ask for this from time to time. Because this keeps coming up (and since we have a non-antialiased `.decimate` for epochs+evoked), I'm okay with adding one that just takes every Nth sample as long as we state clearly that bad things can happen when you do this.

cbrnr, 2022-06-21:
OK! Then instead of adding a parameter to `resample()`, it is better to add a new `decimate()` method as you suggested.

umair-hassan, 2022-12-26:
Ping, just checking for an update here.

jasmainak, 2022-12-26:
see related issue: https://github.com/mne-tools/mne-python/issues/10447