mne-tools/mne-python #10798: Issue reading 4DNI bti MEG file containing EEG channels with labels starting with A

bcolombet, 2022-06-21:
The MEG data file contains EEG channels labeled AF3, AF4, AF7, AF8 .

MNE considers this channels as MEG channel not EEG and raises the following error:

````
Reading 4D PDF file e:\data\mne\2\c,rfDC...
Missing values BTI dev->head transform. Replacing with identity matrix.
Creating Neuromag info structure ...
... Setting channel info structure.
... putting coil transforms in Neuromag coordinates
... Reading digitization points from e:\data\mne\2\hs_file
Currently direct inclusion of 4D weight tables is not supported. For critical use cases please take into account the MNE command "mne_create_comp_data" to include weights as printed out by the 4D "print_table" routine.
Traceback (most recent call last):
  File "C:\Users\bruno\PycharmProjects\mne_debug\main.py", line 25, in <module>
    raw = mne.io.read_raw_bti(fname, rename_channels=False, preload=False)
  File "<decorator-gen-222>", line 12, in read_raw_bti
  File "C:\Users\bruno\PycharmProjects\mne_debug\venv\lib\site-packages\mne\io\bti\bti.py", line 1287, in read_raw_bti
    return RawBTi(pdf_fname, config_fname=config_fname,
  File "<decorator-gen-221>", line 12, in __init__
  File "C:\Users\bruno\PycharmProjects\mne_debug\venv\lib\site-packages\mne\io\bti\bti.py", line 950, in __init__
    info, bti_info = _get_bti_info(
  File "C:\Users\bruno\PycharmProjects\mne_debug\venv\lib\site-packages\mne\io\bti\bti.py", line 1223, in _get_bti_info
    info._check_consistency()
  File "C:\Users\bruno\PycharmProjects\mne_debug\venv\lib\site-packages\mne\io\meas_info.py", line 1101, in _check_consistency
    raise TypeError(
TypeError: Bad info: info["chs"][274]["loc"] must be ndarray with 12 elements, got None
````

When the channels are renamed  FA3, FA4, FA7, FA8  there is no problem to read the data file.

agramfort, 2022-06-22:
@bcolombet thx for the bug report. Can you test with the main branch to see if the bug is still there? If so can you share a file privately with me? :pray:

bcolombet, 2022-06-22:
@agramfort Hi Alexandre, I can share two data files recorded yesterday on the MEG system:
One contains EEG with AF4, ... channels  and raises the error, on the other we renamed AF4 to FA4, ...  and it passed the consistency check routine.
The two runs are available here:
https://amubox.univ-amu.fr/s/4KrdaPSTjBBCTox

agramfort, 2022-06-23:
@bcolombet the bug has been fixed in current development version.

```
mne.io.read_raw_bti('c,rfDC')
<ipython-input-13-978bdf0e35d4>:1: RuntimeWarning: Channel names are not unique, found duplicates for: {'TP7'}. Applying running numbers for duplicates.
  mne.io.read_raw_bti('c,rfDC')
Out[13]: <RawBTi | c,rfDC, 339 x 249487 (122.6 s), ~622 kB, data not loaded>
```

you need to wait for our new 1.1 imminent release or use main for now

bcolombet, 2022-06-23:
@agramfort thank you, I will warn the user to update mne-python to process the data acquired here.

About TP7, yes,  we made a mistake preparing the EEG labels for the recording test: we put TP7 twice , hence the warning.
So this 'bug' is normal :)

agramfort, 2022-06-23:
all good

closed 2022-06-23 by agramfort