mne-tools/mne-python #10806: slow CI docs build

drammock, 2022-06-23:
Yesterday's overnight doc build timed out. A quick scan of the log output shows this:

```
generating gallery for auto_tutorials/clinical... [ 25%] 10_ieeg_localize.py
Clean 2539.8 s : 2.05 GB
Clean 2712.5 s : 2.04 GB
```
https://app.circleci.com/pipelines/github/mne-tools/mne-python/14833/workflows/ee1d9eb6-039a-4178-8395-88441df5bafb/jobs/45868?invite=true#step-139-400

IDK when the time/memory tracking was added, but if I'm interpreting it correctly it seems to mean that at least 172 seconds were spent on rendering that tutorial. @alexrockhill do you have time look into this?  I'd start with a local doc build with `PATTERN` to see how long that tutorial takes locally, then examine recent changes to it to see what might be making it so slow.

alexrockhill, 2022-06-23:
Huh, the only thing that has been changed lately is that the video was added. The dev version up right now says it takes 2 minutes and 47 seconds to execute. There are a lot of computations and components, I think it's going to take close to two minutes. I know @larsoner, optimized it most recently and I thought he got it down below 90 seconds when he did that if I'm not mistaken. Is 172 seconds causing the timeout or it's just the longest-executing tutorial?

alexrockhill, 2022-06-23:
I remember something about this being about a paid circle version and discussions related to if we can keep our circle run under 10 minutes and deciding no and that it would need to be paid for. I'm happy to look into optimizing the ieeg tutorial but it's not clear to me why it's essential for the docs building in time, aren't we supposed to have more than ten minutes?

drammock, 2022-06-23:
> Is 172 seconds causing the timeout or it's just the longest-executing tutorial?

I just scanned through the timings and found what looked like the biggest jump (the "longest files" summary didn't print, because the build job didn't finish). It's likely that this tutorial is not *the sole cause* of the timeout, but if it's taking nearly double what it was recently then that's worth looking into IMO, even if it doesn't "solve" the larger issue.

alexrockhill, 2022-06-23:
I'm happy to look into it, but just checking that I think the larger issue is why we have a ten minute timeout when I thought we fixed that

closed 2023-01-17 by drammock