mne-tools/mne-python #10811: update path handling in examples/inverse/dics_source_power.py

drammock, 2022-06-23:
all of our `data_path` functions now return `pathlib.Path` objects, which allows things like the following:

https://github.com/mne-tools/mne-python/blob/37a9e5587443109515a85cdc4396af4e014ecf82/examples/datasets/brainstorm_data.py#L32-L35

Compare this to the old pattern (using `os.path.join`, often imported as `op.join`):

```py
raw_path = os.path.join(data_path, 'MEG', 'bst_raw',
                        'subj001_somatosensory_20111109_01_AUX-f.ds')
```

or the *really* old pattern (using the `+` operator on plain strings):

```py
raw_path = (data_path + '/MEG/bst_raw/' +
            'subj001_somatosensory_20111109_01_AUX-f.ds')
```

Update the paths in the file mentioned in the issue title to use the new pathlib-style, which is considered the modern, best-practice way to handle file paths.

scott-huberty, 2022-06-24:
This is awesome, I love the `pathlib` module. Happy to work on this next week.

drammock, 2022-06-24:
Great! Everyone will do one of these pathlib issues as their "warmup" on the first day 🙂

closed 2022-06-27 by drammock