mne-tools/mne-python #10819: [BUG] Scalp surface is not loaded for `coreg` GUI is fiducials exist (and looks bad/is unreadable in notebook)

alexrockhill, 2022-06-23:
When I `rm /Users/alexrockhill/mne_data/MNE-sample-data/subjects/sample/bem/sample-fiducials.fif` and then launch the `coreg` GUI with

```
import os.path as op
import mne

data_path = mne.datasets.sample.data_path()
subjects_dir = op.join(data_path, 'subjects')

gui = mne.gui.coregistration(subject='sample', subjects_dir=subjects_dir)
```

it works great. But if the fiducials exist before it's launched I get no scalp surface:

[image]

Also, you can't read anything. I know I can change my Jupyter Notebook zoom settings but I think this is an issue, it should look at least okay so that you can read things with the default settings on a pretty standard machine (below). I think the fontsizes should be way smaller so that the words don't get cut off.

```
Platform:         macOS-11.6.6-x86_64-i386-64bit
Python:           3.9.10 | packaged by conda-forge | (main, Feb  1 2022, 21:28:27)  [Clang 11.1.0 ]
Executable:       /Users/alexrockhill/software/anaconda3/envs/mne/bin/python3.9
CPU:              i386: 4 cores
Memory:           8.0 GB

mne:              1.1.dev0
numpy:            1.21.5 {blas=openblas, lapack=openblas}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=Qt5Agg}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.0
dipy:             1.5.0dev
cupy:             Not found
pandas:           1.4.1
pyvista:          0.33.2 {OpenGL 4.1 INTEL-16.5.11 via Intel(R) Iris(TM) Graphics 6100}
pyvistaqt:        0.7.0
ipyvtklink:       0.2.2
vtk:              9.0.3
qtpy:             2.0.1 {PyQt5=5.12.9}
ipympl:           Not found
pyqtgraph:        0.12.3
pooch:            v1.6.0

mne_bids:         0.11.dev0
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.3.0
mne_connectivity: 0.3dev0
mne_icalabel:     0.2dev0
```

alexrockhill, 2022-06-30:
It also doesn't look like `matplotlib` works in notebook, I get this:

[image]

When, with `pyvistaqt` I get this:

[image]

Obviously, the fact that the notebook GUI goes way off the screen so you can't see half of it is a pretty big issue too but I also thing the canvas backend is broken, when you display locally, you just get text (e.g. `<mne.viz.backends._notebook._IpyMplCanvas at 0x1115a0dc0>`).

Code to reproduce:

```
import os.path as op
import mne

data_path = mne.datasets.sample.data_path()
subjects_dir = op.join(data_path, 'subjects')
sample_dir = op.join(data_path, 'MEG', 'sample')

# gui = mne.gui.coregistration(subject='sample', subjects_dir=subjects_dir)

stc = mne.read_source_estimate(op.join(sample_dir, 'sample_audvis-meg'))
stc.crop(0.09, 0.1)
stc.plot(subject='sample', subjects_dir=subjects_dir)
brain = mne.viz.Brain(subject_id='sample', subjects_dir=subjects_dir,
                      show_toolbar=True)
brain.add_data(stc.lh_data, hemi='lh', vertices=stc.lh_vertno)
brain.add_data(stc.rh_data, hemi='rh', vertices=stc.rh_vertno)
```

larsoner, 2022-06-30:
Indeed the notebook has some bugs including the image display. Sometimes it gets deformed, too. See the `object_fit` stuff in 
https://github.com/mne-tools/mne-python/issues/8833 and also https://github.com/mne-tools/mne-python/issues/8704

larsoner, 2022-10-21:
I think this hopefully has been fixed by the full-screen antialiasing workarounds we have in place for macOS and Mesa

closed 2022-10-21 by larsoner