mne-tools/mne-python #10823: Add ref channel aliases for `RawMFF` (EGI), to `mne.io.constants.CHANNEL_LOC_ALIASES`

scott-huberty, 2022-06-24:
#### Describe the new feature or enhancement
*Please provide a clear and concise description of what you want to add or
change.*

`mne.io.raw.set_montage` includes a parameter `match_alias`, which searches `mne.io.constants.CHANNEL_LOC_ALIASES` for known channel aliases, to match unrecognized channel location names to their known aliases.

In EGI EEG files (RawMFF), the REF channel is given the last channel name (i.e. 'E129' for the GSN-Hydrocel-129, and 'E257' for GSN-Hydrocel-257). However in the MNE-python standard montages for both these nets, the REF channel is named 'Cz', which makes sense as that is what we colloquially call the REF channel.

This discrepancy has caused errors for me when setting the standard montage, and seems to [confuse other users too](https://mne.discourse.group/t/what-eeg-montage-to-use-for-257-channels/5093/11).

#### Describe your proposed implementation

 I propose to add 'E129' and 'E257'  to `mne.io.constants.CHANNEL_LOC_ALIASES` so that `raw.set_montage` can handle the discrepancy in the REF channel name. @drammock this is something that I can work on during the code sprint, and hopefully should be pretty simple.

mscheltienne, 2022-06-25:
I loaded and added a montage to a 257 channels recording with the EGI software last week, and IIRC I tried the `"EGI_256"` build-in montage that lacks the reference. 

I notice now that there are `GSN-HydroCel` montages: `GSN-HydroCel-128`, `GSN-HydroCel-129`, `GSN-HydroCel-256`, `GSN-HydroCel-257`, ... It looks to me like those are duplicates or subset without the reference. **If this is the case**, maybe it would be worth cleaning a bit those default montages, at least by deprecating the duplicates (removing them from the documented list of build-in montages) and keeping only one build-in montage per cap.

scott-huberty, 2022-07-21:
now that #10898 is merged to main I will return to this. Since the reference channel for EGI nets will now be read in as `VREF`, but the reference channel is named `Cz` in the standard montage, I will add this alias to `mne.io.constants.CHANNEL_LOC_ALIASES`. 

Unless anyone thinks the correct fix would be to directly edit the default EGI montages so that the reference channel is called `VREF`.

larsoner, 2022-07-26:
> I will add this alias to mne.io.constants.CHANNEL_LOC_ALIASES.

My worry is that `Cz` is a common, non-ref name in many montages so I'm not sure if something will break. For example would this cause other montages like `standard_1020` to be read in with `Cz` set as the ref, even if it wasn't? I don't know.

If this is the case, we might need to make `CHANNEL_LOC_ALIASES` somehow montage-specific :(

drammock, 2022-07-26:
If the alias changes `VREF` to `Cz` then that's fine, right?  I think that's what would be needed here

larsoner, 2022-07-26:
Yes if that's what happens if sounds good. I'm not too familiar with this code so my speculation above might not make any sense :)

scott-huberty, 2022-07-26:
Yes, the alias maps `VREF` to `Cz` so channels already named `Cz` in `standard_1020` shouldn't be affected, since  As far as I'm aware, `VREF` doesn't appear in `standard_1020` ; )

I have made the changes on my fork and ran some tests locally, I can submit a PR now. If my attempt at making it easier to set a standard EGI montage isn't the best approach, just let me know!

larsoner, 2022-07-26:
Yes PR welcome!

PR #10962 (2022-07-27): Add channel alias to `mne.io.constants.CHANNEL_LOC_ALIASES`
closes #10823 

adds alias for EGI ref channel to `mne.io.constants.CHANNEL_LOC_ALIASES`, since there are currently discrepancies between the ref channel name in EGI files (`VREF` / `Vertex Reference`), and the ref channel name in the EGI standard montages (`Cz`)

This allows the user to set a standard montage without raising an error, for example by calling `raw.set_montage('GSN-HydroCel-129', match_alias=True)`
closed 2022-07-27 by drammock