mne-tools/mne-python #10828: Conda installation hangs forever

wvxvw, 2022-06-26:
#### Describe the bug
Very similar to https://github.com/mne-tools/mne-python/issues/10014, however, I'm trying to install several other packages as well. If I remove `mne` from the list, I'm able to install them (also, slow, but `conda` is never fast...). With `mne` included, I waited for about a day give or take a few hours, and it was still at 100% CPU "solving" something. Obviously, this is also a `conda` bug, but `conda` is beyond repair, so, unfortunately, others need to work around its problems.

The way I discovered the problem was by running `conda build` of a project that has a dependency on `mne`. So, unfortunately, I cannot use `mamba`, because `boa` is still in experimental stage. However, further investigation showed that:

``` sh
conda install -y --strict-channel-priority --override-channels -c conda-forge "python==3.8" "conda==4.13.0" "pandas" "matplotlib" "scipy" "scikit-learn" "h5py" "pyxdf" "mne" "textdistance"
```
exhibits the same behavior.

This happens on Linux. I don't think that flavor or kernel version etc. are important, but will provide those if you think they are.

#### Expected results
The library installs in reasonable amount of time (at least under one hour!)

#### Actual results
No traceback. Also, `conda` swallows `pdb` errors, so, it's impossible to debug w/o much setup. When interrupted, the install doesn't show traceback.

#### Additional information
Impossible to run `mne.sys_info()` since it fails to install.

wvxvw, 2022-06-26:
It seems like other requirements don't matter. Simply installing `mne` hangs forever, i.e.:

``` sh
conda install -y --strict-channel-priority --override-channels -c conda-forge mne
```

is enough to reproduce the problem.

wvxvw, 2022-06-26:
Further experiments showed that `mne-base` can be installed in the same context.

agramfort, 2022-06-26:
what do you mean by context? current environment?

> Message ID: ***@***.***>
>

wvxvw, 2022-06-26:
I mean that it worked with Python 3.8 and `conda` 4.13.0, also all the other packages listed in the first post can be installed if I replace `mne` with `mne-base`.

agramfort, 2022-06-26:
ok but mne-base is very light weight. mne requires a bunch of stuff (3D
libs, GUIs etc)

can you try on a fresh env or using mamba in you local environment?

> Message ID: ***@***.***>
>

wvxvw, 2022-06-26:
I have to leave now, but I'll loop back to this tonight, so, hopefully will have more results.

Luckily for me, I only really needed the stuff from `mne-base` anyways, just didn't know it was a thing.

wvxvw, 2022-06-26:
`mamba` has no problems installing `mne` with all the same requirements.

agramfort, 2022-06-26:
so let's use mamba ;)

> Message ID: ***@***.***>
>

drammock, 2022-06-30:
ok so if all you really need is `mne-base` and it works with the version of `conda` that you want, then this can be closed, yes?  (If not we may need to close anyway as unfortunately there's little we can do to make conda's solver work better)

wvxvw, 2022-07-01:
Maybe drop a line to Anaconda's people? I'm too tired of reminding them how `conda` is broken. They probably ignore this coming from the same person anyways.

Or, at the minimum, you could write in installation instructions to not use `conda`, since it doesn't work. Would save people some headache.

cbrnr, 2022-07-01:
Since you seem to be very unhappy with conda, have you tried using pip? This is the official Python package manager, and I have found pip to work extremely well. I do not use conda/mamba for the same reasons you seem to have, so maybe pip is worth a try?

larsoner, 2022-10-21:
Nowadays we mostly advocate for using our installers. Closing as a (mostly) upstream issue

closed 2022-10-21 by larsoner