mne-tools/mne-python #10836: split raw file error in decoding_rasa_sgskip.py

CarinaFo, 2022-06-27:


drammock, 2022-06-27:
I've reproduced this locally on current `main`

drammock, 2022-06-27:
it seems that the visual_92 data files have a `FIFF_REF_FILE_NUM` tag but don't actually have subsequent files (?).  This dataset is only actually used in 2 places in our docs, and both are `_sgskip` files so they never actually get executed/tested. Here are the lines where the tag gets matched and triggers `next_fname` to not be `None`:

https://github.com/mne-tools/mne-python/blob/23a8da684a4debe87b8de99dda6fba399ac22063/mne/io/open.py#L73-L91

I'm not sure how to dig deeper; I don't know much about the origin of that dataset (or how things like split files are encoded in the FIFF standard)

drammock, 2022-06-29:
this wasn't actually fixed yet, as the linked PR wasn't merged. reopening.