mne-tools/mne-python #10848: Get KeyError: 'float', when trying to plot EEG data from .mff file

Kiranraj227, 2022-06-27:
#### Describe the bug
*When I try to plot the raw data, I get this error: KeyError: 'float'*

#### Steps to reproduce
import mne
import matplotlib

matplotlib.use('Qt5Agg')
fname = r"C:\Users\HP\Desktop\MD.mff"
raw = mne.io.read_raw_egi(fname)
raw.crop(tmax=60).load_data()
#error occurs when running the line below
raw.plot(duration=20, scalings=20e-6)

#### Expected results
*Expect to plot EEG channels*

#### Actual results
Using qt as 2D backend.
Opening raw-browser...
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Input In [3], in <cell line: 1>()
----> 1 raw.plot(duration=20, scalings=20e-6)

File ~\anaconda3\envs\mne-light\lib\site-packages\mne\io\base.py:1556, in BaseRaw.plot(self, events, duration, start, n_channels, bgcolor, color, bad_color, event_color, scalings, remove_dc, order, show_options, title, show, block, highpass, lowpass, filtorder, clipping, show_first_samp, proj, group_by, butterfly, decim, noise_cov, event_id, show_scrollbars, show_scalebars, time_format, precompute, use_opengl, theme, verbose)
   1546 @copy_function_doc_to_method_doc(plot_raw)
   1547 def plot(self, events=None, duration=10.0, start=0.0, n_channels=20,
   1548          bgcolor='w', color=None, bad_color='lightgray',
   (...)
   1554          show_scrollbars=True, show_scalebars=True, time_format='float',
   1555          precompute=None, use_opengl=None, *, theme=None, verbose=None):
-> 1556     return plot_raw(self, events, duration, start, n_channels, bgcolor,
   1557                     color, bad_color, event_color, scalings, remove_dc,
   1558                     order, show_options, title, show, block, highpass,
   1559                     lowpass, filtorder, clipping, show_first_samp,
   1560                     proj, group_by, butterfly, decim, noise_cov=noise_cov,
   1561                     event_id=event_id, show_scrollbars=show_scrollbars,
   1562                     show_scalebars=show_scalebars, time_format=time_format,
   1563                     precompute=precompute, use_opengl=use_opengl,
   1564                     theme=theme, verbose=verbose)

File <decorator-gen-182>:12, in plot_raw(raw, events, duration, start, n_channels, bgcolor, color, bad_color, event_color, scalings, remove_dc, order, show_options, title, show, block, highpass, lowpass, filtorder, clipping, show_first_samp, proj, group_by, butterfly, decim, noise_cov, event_id, show_scrollbars, show_scalebars, time_format, precompute, use_opengl, theme, verbose)

File ~\anaconda3\envs\mne-light\lib\site-packages\mne\viz\raw.py:356, in plot_raw(raw, events, duration, start, n_channels, bgcolor, color, bad_color, event_color, scalings, remove_dc, order, show_options, title, show, block, highpass, lowpass, filtorder, clipping, show_first_samp, proj, group_by, butterfly, decim, noise_cov, event_id, show_scrollbars, show_scalebars, time_format, precompute, use_opengl, theme, verbose)
    305 precompute = _handle_precompute(precompute)
    306 params = dict(inst=raw,
    307               info=info,
    308               # channels and channel order
   (...)
    353               use_opengl=use_opengl,
    354               theme=theme)
--> 356 fig = _get_browser(show=show, block=block, **params)
    358 return fig

File ~\anaconda3\envs\mne-light\lib\site-packages\mne\viz\_figure.py:654, in _get_browser(show, block, **kwargs)
    651             return fig
    653 # Initialize Browser
--> 654 fig = backend._init_browser(**kwargs)
    655 _show_browser(show=show, block=block, fig=fig)
    657 return fig

File ~\anaconda3\envs\mne-light\lib\site-packages\mne_qt_browser\_pg_figure.py:4702, in _init_browser(**kwargs)
   4700 if 'splash' in app_kwargs:
   4701     kwargs['splash'] = out[1]  # returned as second element
-> 4702 browser = MNEQtBrowser(**kwargs)
   4704 return browser

File ~\anaconda3\envs\mne-light\lib\site-packages\mne_qt_browser\_pg_figure.py:153, in _safe_splash.<locals>.func(self, *args, **kwargs)
    150 @functools.wraps(meth)
    151 def func(self, *args, **kwargs):
    152     try:
--> 153         meth(self, *args, **kwargs)
    154     finally:
    155         try:

File ~\anaconda3\envs\mne-light\lib\site-packages\mne_qt_browser\_pg_figure.py:2805, in MNEQtBrowser.__init__(self, **kwargs)
   2802 self.mne.viewbox = RawViewBox(self)
   2804 # Start precomputing if enabled
-> 2805 self._init_precompute()
   2807 # Parameters for overviewbar
   2808 self.mne.overview_mode = getattr(self.mne, 'overview_mode', 'channels')

File ~\anaconda3\envs\mne-light\lib\site-packages\mne_qt_browser\_pg_figure.py:3816, in MNEQtBrowser._init_precompute(self)
   3813 gc.collect()
   3815 if self.mne.precompute == 'auto':
-> 3816     self.mne.enable_precompute = self._check_space_for_precompute()
   3817 elif isinstance(self.mne.precompute, bool):
   3818     self.mne.enable_precompute = self.mne.precompute

File ~\anaconda3\envs\mne-light\lib\site-packages\mne_qt_browser\_pg_figure.py:3863, in MNEQtBrowser._check_space_for_precompute(self)
   3859     fmt = getattr(self.mne.inst, 'orig_format', 'single')
   3860     # Apply size change to 64-bit float in memory
   3861     # (* 2 because when loading data will be loaded into a copy
   3862     # of self.mne.inst._data to apply processing.
-> 3863     expected_ram = disk_space * fmt_multipliers[fmt] * 2
   3864 else:
   3865     expected_ram = sys.getsizeof(self.mne.inst._data)

KeyError: 'float'

#### Additional information
*Platform:         Windows-10-10.0.19044-SP0
Python:           3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\Users\HP\anaconda3\envs\mne-light\python.exe
CPU:              Intel64 Family 6 Model 61 Stepping 4, GenuineIntel: 4 cores
Memory:           15.9 GB

mne:              1.0.3
numpy:            1.22.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.1
matplotlib:       3.5.2 {backend=Qt5Agg}

sklearn:          1.1.1
numba:            0.55.2
nibabel:          4.0.1
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.3
pyvista:          0.34.1 {OpenGL 4.4.0 - Build 20.19.15.4549 via Intel(R) HD Graphics 5500}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
PyQt5:            5.12.3
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.3.1
mne_connectivity: Not found*

Kiranraj227, 2022-06-28:
When I checked the file that the error message was referring to (mne_qt_browser_pg_figure.py:3863,), I noticed the dictionary for fmt_multipliers did not contain float as one of its options as shown below:
```
# Determine expected RAM space based on orig_format
                fmt_multipliers = {'double': 1,
                                   'single': 2, 
                                   'int': 2,
                                   'short': 4}

                # Epochs and ICA don't have this attribute, assume single
                # on disk
                fmt = getattr(self.mne.inst, 'orig_format', 'single')
                # Apply size change to 64-bit float in memory
                # (* 2 because when loading data will be loaded into a copy
                # of self.mne.inst._data to apply processing.
                expected_ram = disk_space * fmt_multipliers[fmt] * 2
            else:
                expected_ram = sys.getsizeof(self.mne.inst._data)
```
Therefore, it raises the KeyError: 'float'

Please note that for my dataset (MD.mff), the orig_format = 'float'

Please advise a workaround for this. The weird thing is when running the same dataset on another laptop it works fine and doesn't trigger this error.

mscheltienne, 2022-06-28:
I can reproduce locally with a `.mff` file, and I ran into this when trying to add EGI as a manufacturer for `mne-bids`.
I think the reader should be improved to load an `orig_format` of `single` or `double`.

As a workaround, you can set the format to `double` with `raw.orig_format = "double"`. Then the plot with the qt browser works.

Kiranraj227, 2022-06-28:
It finally works! I also edited the library by adding the double to the dict. Thanks a lot for your help. Spent the whole night with this error.

Kiranraj227, 2022-06-28:
```
    def _check_space_for_precompute(self):
        try:
            import psutil
        except ImportError:
            logger.info('Free RAM space could not be determined because'
                        '"psutil" is not installed. '
                        'Setting precompute to False.')
            return False
        else:
            if self.mne.is_epochs:
                files = [self.mne.inst.filename]
            else:
                files = self.mne.inst.filenames
            if files[0] is not None:
                # Get disk-space of raw-file(s)
                disk_space = 0
                for fn in files:
                    disk_space += getsize(fn)

                # Determine expected RAM space based on orig_format
                fmt_multipliers = {'double': 1,
                                   'float': 2, ### added 'float' to the dict
                                   'single': 2,
                                   'int': 2,
                                   'short': 4}

                # Epochs and ICA don't have this attribute, assume single
                # on disk
                fmt = getattr(self.mne.inst, 'orig_format', 'float') #'single' to 'float'
                # Apply size change to 64-bit float in memory
                # (* 2 because when loading data will be loaded into a copy
                # of self.mne.inst._data to apply processing.
                expected_ram = disk_space * fmt_multipliers[fmt] * 2
            else:
                expected_ram = sys.getsizeof(self.mne.inst._data)

            # Get available RAM
            free_ram = psutil.virtual_memory().free

            expected_ram_str = sizeof_fmt(expected_ram)
            free_ram_str = sizeof_fmt(free_ram)
            left_ram_str = sizeof_fmt(free_ram - expected_ram)

            if expected_ram < free_ram:
                logger.debug('The data precomputed for visualization takes '
                             f'{expected_ram_str} with {left_ram_str} of '
                             f'RAM left.')
                return True
            else:
                logger.debug(f'The precomputed data with {expected_ram_str} '
                             f'will surpass your current {free_ram_str} '
                             f'of free RAM.\n'
                             'Thus precompute will be set to False.\n'
                             '(If you want to precompute nevertheless, '
                             'then set precompute to True instead of "auto")')
                return False``
```

mscheltienne, 2022-06-28:
Could you leave this issue open? It would be good to fix it for everyone rather than patch it every time.

PR #10851 (2022-06-30): MRG: Change orig_format for EGI, BrainVision ASCII and artemis123 recordings to "single"
Closes #10848.

I'm, 99% sure the `dtype` for EGI recordings is `float32`, and that seems coherent with the code in MNE (`'<f4'` and in `mffpy`). Instead of setting the `.orig_format` attribute to `"float"`, it now sets the attribute to `"single"`.
EDIT: I've checked the specification, it is float32.

The QT browser is compatible with `"single", "double", "int", and "short"`: https://github.com/mne-tools/mne-qt-browser/blob/d9c23703658abfc56113cdaff8470952bf3100b1/mne_qt_browser/_pg_figure.py#L3853-L3856

 [... cut]
closed 2022-06-30 by larsoner