mne-tools/mne-python #10852: test_search_light fails on CI pre jobs

drammock, 2022-06-28:
possibly an sklearn behavior change?  Haven't dug any deeper yet.

Failing on linux [here](https://github.com/mne-tools/mne-python/runs/7088858139?check_suite_focus=true#step:13:3625) and windows [here](https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=20371&view=logs&j=305851a9-a7bb-55db-0042-7e2b6f48aa1c&t=77eb5bf7-3205-5b54-06c6-183c8195d846&l=3199)

```
______________________________ test_search_light _______________________________
mne/decoding/tests/test_search_light.py:150: in test_search_light
    pipe.fit(X, y)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/pipeline.py:378: in fit
    Xt = self._fit(X, y, **fit_params_steps)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/pipeline.py:336: in _fit
    X, fitted_transformer = fit_transform_one_cached(
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/memory.py:349: in __call__
returnself.func(*args, **kwargs)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/pipeline.py:870: in _fit_transform_one
    res = transformer.fit_transform(X, y, **fit_params)
mne/decoding/search_light.py:117: in fit_transform
returnself.fit(X, y, **fit_params).transform(X)
mne/decoding/search_light.py:80: in fit
    estimators = parallel(
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/parallel.py:1043: in __call__
ifself.dispatch_one_batch(iterator):
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/parallel.py:861: in dispatch_one_batch
self._dispatch(tasks)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/parallel.py:779: in _dispatch
    job = self._backend.apply_async(batch, callback=cb)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/_parallel_backends.py:208: in apply_async
    result = ImmediateResult(func)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/_parallel_backends.py:572: in __init__
self.results = batch()
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/parallel.py:262: in __call__
return [func(*args, **kwargs)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/joblib/parallel.py:262: in <listcomp>
return [func(*args, **kwargs)
mne/decoding/search_light.py:318: in _sl_fit
    est.fit(X[..., ii], y, **fit_params)
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/linear_model/_logistic.py:1098: in fit
self._validate_params()
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/base.py:613: in _validate_params
    validate_parameter_constraints(
/opt/hostedtoolcache/Python/3.10.5/x64/lib/python3.10/site-packages/sklearn/utils/_param_validation.py:47: in validate_parameter_constraints
raiseValueError(
E   ValueError: The parameter constraints ['penalty', 'dual', 'tol', 'C', 'fit_intercept', 'intercept_scaling', 'class_weight', 'random_state', 'solver', 'max_iter', 'multi_class', 'verbose', 'warm_start', 'n_jobs', 'l1_ratio'] do not match the parameters to validate [].
```

agramfort, 2022-06-28:
yes sklearn pre was just released I think.... I need to look into this

> Message ID: ***@***.***>
>

drammock, 2022-07-05:
this was fixed by #10889

closed 2022-07-05 by drammock