mne-tools/mne-python #10867: Unexpected behavior when Epoching data on cropped Raw with .set_annotations

matthiasdold, 2022-06-29:
#### Describe the bug
I am cropping a `raw` object to exclude the first data (not containing any trials which are relevant for me). Later I need to work on the data as an array (plain `numpy`) and bring everything back to a new raw `nraw` object (type: mne.io.array.array.RawArray), including setting the annotations from the cropped but otherwise not modified `raw`. 

Now the data and the events are as expected, but once I epoch the `nraw`, a few epochs at the end get dropped and the epochs no longer match.

#### Steps to reproduce
```python
import mne
from mne.io import concatenate_raws, read_raw_edf
from mne.datasets import eegbci

import matplotlib.pyplot as plt

subject = 1
runs = [6, 10, 14]
raw_fnames = eegbci.load_data(subject, runs)
raws = [read_raw_edf(f, preload=True) for f in raw_fnames]
raw = concatenate_raws(raws)

# just crop of a bit of time
not_cropped_raw = raw.copy()
raw.crop(20, None)

# now create a new raw object and set the annotations
nraw = mne.io.RawArray(raw.get_data(), raw.info)
nraw.set_annotations(raw.annotations)

# Data and time-stamps perfectly agree
assert (raw.get_data() == nraw.get_data()).all(), "Data missmatch"
assert (raw.times == nraw.times).all(), "Time missmatch"

ev, evid = mne.events_from_annotations(raw)
nev, nevid = mne.events_from_annotations(raw)
assert (ev == nev).all(), "Events missmatch"

# The problem starts when creating the Epochs -> take same events as they are equal anyways, see assert above
epo = mne.Epochs(raw, ev)
nepo = mne.Epochs(nraw, ev)

epo.load_data()
nepo.load_data()

# during the drop step, epochs from the end are dropped -> 'TOO_SHORT'
epo.drop_bad()
nepo.drop_bad()
print(nepo.drop_log[-5:])

# But also the first epochs do not agree, see eg. epo[10]
it = 10
chans = ['Cz..']
clr = ['#f00', '#00f', '#0f0']

epo_data = epo.copy().pick_channels(chans).get_data()[it, 0, :]
nepo_data = nepo.copy().pick_channels(chans).get_data()[it, 0, :]

fig, axs = plt.subplots(1, 1)
axs.plot(epo_data, color=clr[0], label='Cropped')
axs.plot(nepo_data, color=clr[1], label='CroppedAndAnnotationSet')
axs.legend()
plt.show()
```
[image]

#### Expected results
Both epochs `epo` and `nepo` in the minimal example should agree.

#### Actual results
Epochs in `nepo` are not matching `epo`

#### Additional information
I suspect that there is an issue with some time off-set while the epochs are created. I will try to find the root cause for the miss-match, but maybe some knows were things are going wrong already?

agramfort, 2022-06-30:
here is a code that works as expected:

```python
import mne
from mne.io import concatenate_raws, read_raw_edf
from mne.datasets import eegbci

import matplotlib.pyplot as plt

subject = 1
# runs = [6, 10, 14]
runs = [6]
raw_fnames = eegbci.load_data(subject, runs)
raws = [read_raw_edf(f, preload=True) for f in raw_fnames]
raw = concatenate_raws(raws)

# just crop of a bit of time
not_cropped_raw = raw.copy()
raw.crop(20, None)

# now create a new raw object and set the annotations
nraw = mne.io.RawArray(raw.get_data(), raw.info, first_samp=raw.first_samp)
nraw.set_annotations(raw.annotations)

# Data and time-stamps perfectly agree
assert (raw.get_data() == nraw.get_data()).all(), "Data missmatch"
assert (raw.times == nraw.times).all(), "Time missmatch"

ev, evid = mne.events_from_annotations(raw)
nev, nevid = mne.events_from_annotations(nraw)
assert (ev == nev).all(), "Events missmatch"

# The problem starts when creating the Epochs -> take same events as they are equal anyways, see assert above
epo = mne.Epochs(raw, ev)
nepo = mne.Epochs(nraw, ev)

epo.load_data()
nepo.load_data()

# during the drop step, epochs from the end are dropped -> 'TOO_SHORT'
epo.drop_bad()
nepo.drop_bad()
print(nepo.drop_log[-5:])

# But also the first epochs do not agree, see eg. epo[10]
it = 10
chans = ['Cz..']
clr = ['#f00', '#00f', '#0f0']

epo_data = epo.copy().pick_channels(chans).get_data()[it, 0, :]
nepo_data = nepo.copy().pick_channels(chans).get_data()[it, 0, :]

fig, axs = plt.subplots(1, 1)
axs.plot(epo_data, color=clr[0], label='Cropped')
axs.plot(nepo_data, color=clr[1], label='CroppedAndAnnotationSet')
axs.legend()
plt.show()
```

the 2 important lines are 

`nraw = mne.io.RawArray(raw.get_data(), raw.info, first_samp=raw.first_samp)`
and
`nev, nevid = mne.events_from_annotations(nraw)`

matthiasdold, 2022-06-30:
Thank you very much @agramfort! Sorry that I did not spot the `first_samp` kwarg.

I guess this will correctly set `nraw._cropped_samp` or `nraw._first_samps` and `nraw._last_samps` on which the epoching will rely on, right?

Either way, this can be closed of course.

closed 2022-06-30 by matthiasdold

agramfort, 2022-06-30:
>
> I guess this will correctly set nraw._cropped_samp or nraw._first_samps
> and nraw._last_samps on which the epoching will rely on, right?
>
correct

> Message ID: ***@***.***>
>