mne-tools/mne-python #10871: plot_topomap()'s cnorm param should accept a CenteredNorm

hoechenberger, 2022-06-29:
Plotting T statistics on topographic maps with @SophieHerbst, I discovered that `mne.viz.plot_topomap()` only accepts `TwoSlopeNorm` as `cnorm` paramter.

However, I believe for our purpose, [`TwoSlopeNorm`](https://matplotlib.org/stable/api/_as_gen/matplotlib.colors.TwoSlopeNorm.html) is not the best solution:

> Normalize data with a set center.
> Useful when mapping data with an unequal rates of change around a conceptual center, e.g., data that range from -2 to 4, with 0 as the midpoint.

Instead, I believe [`CenteredNorm`](https://matplotlib.org/stable/api/_as_gen/matplotlib.colors.CenteredNorm.html) makes much more sense:

> Normalize symmetrical data around a center (0 by default).
> Unlike [TwoSlopeNorm](https://matplotlib.org/stable/api/_as_gen/matplotlib.colors.TwoSlopeNorm.html#matplotlib.colors.TwoSlopeNorm), [CenteredNorm](https://matplotlib.org/stable/api/_as_gen/matplotlib.colors.CenteredNorm.html#matplotlib.colors.CenteredNorm) applies an equal rate of change around the center.
> Useful when mapping symmetrical data around a conceptual center e.g., data that range from -2 to 4, with 0 as the midpoint, and with equal rates of change around that midpoint.

Hence, I'd like to have us accept `CenteredNorm` as well. Currenly, this is not possible:

https://github.com/mne-tools/mne-python/blob/0452d69396ade5186bfad166aa7c7f608b288867/mne/viz/topomap.py#L800

drammock, 2022-06-29:
probably `TwoSlopeNorm` and `CenteredNorm` have a parent class they both inherit from; if so we could/should use that in the call to `_validate_type`

hoechenberger, 2022-06-30:
Yes they do. Although I almost believe that there are probably very few (no?) use cases for TwoSlopeNorm when plotting topomaps... because it creates "asymmetrical" gradients 🤔

wmvanvliet, 2022-06-30:
ping @cbrnr :)

cbrnr, 2022-06-30:
To support asymmetrical mappings and still have white represent zero. For example, ERD/ERS values range from -1 to e.g. 2.5, but I want 0 to be "in the middle" (white).

cbrnr, 2022-06-30:
I don't understand what they mean with "equal/unequal rates of change around a conceptual center" though.

wmvanvliet, 2022-06-30:
Say the data is from -1 to 2.5, and the colormap is blue --> white ---> red. We would normally want the colormap to be symmetric: -2.5 to be dark blue, 0 to be white and 2.5 to be dark red. I think the `TwoSlopeNorm` will make -1 dark blue, so an assymetric colormap.

wmvanvliet, 2022-06-30:
Aah I see. So, if you just want a symmetric colormap like in my example, normally you would just use `vmin == vmax` and be done. But in the exceptional case where you don't want a symmetric map, this used to be impossible without the `cnorm` parameter. Hence, the doc for `cnorm` mentions `TwoSlopeNorm` because it describes this case.

cbrnr, 2022-06-30:
Exactly!

PR #10880 (2022-06-30): Relax the type checking for the cnorm paramter of plot_topomap
`mne.viz.plot_topomap` takes a `cnorm` parameter to allow one to pass a [`matplotlib.colors.Normalize`](https://matplotlib.org/stable/api/_as_gen/matplotlib.colors.Normalize.html#matplotlib.colors.Normalize) object to control how the colormap is applied to the values. Common use cases is to force the 0 to be at the center of the colormap. Before, only `TwoSlopeNorm` was allowed, but there is no reason not to accept all subclasses of `Normalize` here.

closes #10871