mne-tools/mne-python #10882: FileNotFoundError: The path you specified, "slp01a.dat.lay",does not exist.

soso-maitha, 2022-06-30:
#### Describe the bug
I get the following error when I try to read .dat file
FileNotFoundError: The path you specified, "slp01a.dat.lay",does not exist.

#### Steps to reproduce
import os 
import mne

the data are from MIT-BIH Polysomnographic Database https://physionet.org/content/slpdb/1.0.0/
slp1a_data = mne.io.read_raw_persyst('slp01a.dat')

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
Expected to read raw data from .dat file

#### Actual results
Error FileNotFoundError: The path you specified, "slp01a.dat.lay",does not exist.

I don't know why it is adding ".lay" at the end

#### Additional information
Platform:         Linux-5.4.0-1083-azure-x86_64-with-glibc2.10
Python:           3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:       /anaconda/envs/azureml_py38/bin/python
CPU:              x86_64: 6 cores
Memory:           54.9 GB

mne:              1.0.3
numpy:            1.19.0 {blas=openblas, lapack=openblas}
scipy:            1.5.2
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          0.22.1
numba:            0.53.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.1.5
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            5.15.7
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found

drammock, 2022-06-30:
Are you sure the data come from a Persyst system?  the `.dat` file extension is used by more than one manufacturer.  There doesn't seem to be a `.lay` file, instead there is a `.hea` (header) file.  cc @adam2392 who wrote the Persyst reader.

adam2392, 2022-06-30:
I do not think the data comes from Persyst. So unfortunately no idea. 

If it's a new system, would be nice to have it as a PR

drammock, 2022-06-30:
@soso-maitha can you help by figuring out what the data format is?  I skimmed https://ecg.mit.edu/george/publications/slpdb-pcn-1999.pdf but they don't mention Persyst.

drammock, 2022-06-30:
I've never worked with this format so I'm not sure what else to say. Having a `.dat` extension in no way guarantees that the file contents are a valid persyst file, and the absence of a `.lay` file seems to support the hypothesis that it is in fact something else. On the contrary, if you are able to open the file in some other software (e.g., Persyst's proprietary software, or in Fieldtrip or Brainstorm) then that would indicate that it is our reader function that is too stringent.  I suggest trying that and/or contacting the creators of the file to find out what the format is.

KTibow, 2024-02-20:
Sorry for derailing this thread but anyone know how to parse the ECG files from the same paper? I've tried so much stuff and I have no idea what type of file this is

closed 2024-07-27 by sappelhoff