mne-tools/mne-python #10883: ValueError: invalid literal for int() with base 10: '09:01:17' when trying to read EDF file

soso-maitha, 2022-06-30:
#### Describe the bug
I get the following error when I try to read an EDF file
`ValueError: invalid literal for int() with base 10: '09:01:17'`

#### Steps to reproduce
```
import mne
data = mne.io.read_raw_edf("ucddb009_lifecard.edf",preload = True )
```
This is the database https://physionet.org/content/ucddb/1.0.0/

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Additional information
 output of `mne.sys_info()`.
 
 Platform:         Linux-5.4.0-1083-azure-x86_64-with-glibc2.10
Python:           3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:       /anaconda/envs/azureml_py38/bin/python
CPU:              x86_64: 6 cores
Memory:           54.9 GB

mne:              1.0.3
numpy:            1.19.0 {blas=openblas, lapack=openblas}
scipy:            1.5.2
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          0.22.1
numba:            0.53.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.1.5
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            5.15.7
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found

drammock, 2022-06-30:
This is not a valid EDF file.  Quoting from [the original standard](https://www.edfplus.info/specs/edf.html):

> 8 ascii : starttime of recording (hh.mm.ss) 

and [the EDF+ standard](https://www.edfplus.info/specs/edfplus.html), section 2.1.3:

> 2. The 'startdate' and 'starttime' fields in the header should contain only characters 0-9, and the period (.) as a separator [...]

These files appear to be using a colon separator for times.

You can hack it to work with these files by changing the `.` to `:` on this line:

https://github.com/mne-tools/mne-python/blob/e49525c470b01b04c7270dc054bbfa7600021262/mne/io/edf/edf.py#L656

gaoyangbuchicao, 2022-08-30:
I have to say thanks for your smart "idea" to solve this annoying problem!!!@drammock

larsoner, 2022-10-21:
Closing as I think no changes are needed at our end

closed 2022-10-21 by larsoner