mne-tools/mne-python #10890: 3D plots show as blank scenes (pyvista), no plots in jupyter lab (notebook)

Immiora, 2022-06-30:
This worked just fine in `mne=0.22.0`, but does not in `mne=1.0.3`

Tried installing mne in three different ways:
1) via pip + conda and manually adding latest pyvista, pyvistaqt, vtk, PyQt
2) via the requirements.txt from [here](https://mne.tools/stable/install/advanced.html) and `pip install --user -r requirements.txt`
3) using the packaged installer from [here](https://mne.tools/stable/install/installers.html#installers)

Could not do mamba install because conda hangs on "solving environment"

This is the best I can get with pyvista:
[image]

Running default example from [here](https://mne.tools/stable/auto_tutorials/clinical/30_ecog.html) produces the same result.
All installations result in the same, so I am guessing there must be something off with our hardware/drivers.

Interestingly, simple pyvista examples work just fine:
```
from pyvista import examples
globe = examples.load_globe()
globe.plot()
```
[image]

Here is `pyvista.Report()`:
```
--------------------------------------------------------------------------------
  Date: Thu Jun 30 21:33:17 2022 CEST
                OS : Linux
            CPU(s) : 40
           Machine : x86_64
      Architecture : 64bit
               RAM : 1007.8 GiB
       Environment : IPython
       File system : ext4
        GPU Vendor : Mesa/X.org
      GPU Renderer : llvmpipe (LLVM 11.0.1, 256 bits)
       GPU Version : 4.5 (Core Profile) Mesa 20.3.5
  Python 3.9.2 (default, Feb 28 2021, 17:03:44)  [GCC 10.2.1 20210110]
           pyvista : 0.32.0
               vtk : 9.1.0
             numpy : 1.19.5
           imageio : 2.19.3
           appdirs : 1.4.4
            scooby : 0.5.12
            meshio : 4.4.6
        matplotlib : 3.3.4
         pyvistaqt : 0.9.0
             PyQt5 : 5.15.2
           IPython : 7.20.0
        ipyvtklink : 0.2.2
             scipy : 1.6.0
              tqdm : 4.64.0
--------------------------------------------------------------------------------
```

here is `mne.sys_info()`
```
Platform:         Linux-5.10.0-12-amd64-x86_64-with-glibc2.31
Python:           3.9.2 (default, Feb 28 2021, 17:03:44)  [GCC 10.2.1 20210110]
Executable:       /usr/bin/python3.9
CPU:              : 40 cores
Memory:           1007.8 GB
mne:              1.0.3
numpy:            1.19.5 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.6.0
matplotlib:       3.3.4 {backend=TkAgg}
sklearn:          1.1.1
numba:            0.55.2
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.32.0 {OpenGL 4.5 (Core Profile) Mesa 20.3.5 via llvmpipe (LLVM 11.0.1, 256 bits)}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
PyQt5:            5.15.2
ipympl:           Not found
pooch:            v1.6.0
mne_bids:         0.10
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.3.1
mne_connectivity: Not found
```

Anyone know how to fix this?
Any suggestions are appreciated!

agramfort, 2022-07-01:
can you try with a fresh environment?

something like this:
conda create --name mne_env python=3.9 mne
then activate it with:
conda activate mne_env

Immiora, 2022-07-01:
Hi, thanks for your reply!

Yes, I did clean install with both options 1 and 3. The first one using my existing anaconda. Option 3 installs a fresh separate conda even. The result is the same for all installations.

Immiora, 2022-07-04:
A bit of an update. I am able to plot things correctly if I make calls directly to `pyvista` and `vtk` (avoiding calls to `mne`). 

Any ideas what is going on?

```
import numpy as np
import os
import nibabel
import argparse
import os.path as op
import pyvista as pv
import vtk

def apply_transform(trans, pts, move=True):
    # (still works if calling mne for applying transform and loading surf)
    # apply rotation & scale 
    out_pts = np.dot(pts, trans[:3, :3].T)
    # apply translation
    if move:
        out_pts += trans[:3, 3]
    return out_pts

##
def visualize(subject, freesurfer_dir, coordinates, hemisphere, data_vector=None, output_dir=None):
    # transform coordinates to RAS
    x = nibabel.load(os.path.join(freesurfer_dir, 'mri', 'orig.mgz'))
    vox_coords = np.round(apply_transform(np.linalg.inv(x.affine), coordinates)).astype(int)
    ras_coords = apply_transform(x.header.get_vox2ras_tkr(), vox_coords)
    ras_coords = ras_coords / 1000

    # load freesurfer surface
    brain = 'pial'
    subjects_dir = os.path.dirname(freesurfer_dir)
    brain_fname = op.join(subjects_dir, subject, 'surf', f'{hemisphere}.{brain}')
    surf = nibabel.freesurfer.read_geometry(brain_fname)

    # make surface mesh
    vertices = np.array(surf[0]/1000.)
    triangles = np.array(surf[1])
    triangles = np.c_[np.full(len(triangles), 3), triangles]
    mesh = pv.PolyData(vertices, triangles)

    # plot surface
    plotter = pv.Plotter()
    plotter.add_mesh(mesh, smooth_shading=True, interpolate_before_map=True, opacity=1,
                         style='surface', line_width=1, backface_culling=True, rgba=False)

    # plot electrodes
    sphere = vtk.vtkSphereSource()
    sphere.SetThetaResolution(8)
    sphere.SetPhiResolution(8)
    sphere.Update()
    mesh2 = pv.PolyData(np.array(ras_coords))

    plotter.add_mesh(mesh=mesh2, color=(1,0,0), scalars=data_vector,
                    backface_culling=True, point_size=8, lighting=True,
                    smooth_shading=True, render_points_as_spheres=True,
                     show_scalar_bar=True, scalar_bar_args={'title': 'Statistic'})
    plotter.camera.azimuth = 130 # will depend on hemisphere to visualize
    plotter.camera.elevation = -30 # will depend on hemisphere to visualize
    if output_dir is not None:
        if not os.path.isdir(output_dir): os.makedirs(output_dir)
        img = plotter.show(return_img=True, screenshot=op.join(output_dir, subject + '.png')) # return_cpos, return_img
    else:
        img = plotter.show(return_img=True)
    return img
```

The calling 
```
visualize(subject, fs_dir, coords, hemi, data_vector=vec, output_dir=output_dir)
```
produces the expected plot:
[image]

Is there anything specific in the way `pyvista` is accessed for rendering from `mne` that could cause a blank scene (see comment 1)?

larsoner, 2022-10-21:
Closing under the assumption that this was an antialiasing bug that we fixed (or can be worked around using 3D options)

closed 2022-10-21 by larsoner

claytonjschneider, 2022-10-31:
hi @larsoner sorry to find this just after closing, but I am having the same issue as @Immiora

claytonjschneider, 2022-10-31:
Sike! Using the 3d options line fixed it for me.