mne-tools/mne-python #10899: Problem with visual_92_categories dataset

hoechenberger, 2022-07-02:
Reported at https://mne.discourse.group/t/mne-visual-92-categories-data-sample-subject-0-raw-1-fif-does-not-exist/5204

The dataset consists of 4 split files, but they've been manually renamed.

To fix:
Each file should be read individually, and `on_split_missing='warn'` should be passed. Then the file should be re-saved without it getting split. So we're basically manually splitting the files, as is expected by the Representational Similarity Analysis example.

Alternatative:
Actually the decoding for all runs finishes in basically no time on my computer, so there's no need to do this kind of hocus-pocus in the Representational Similarity Analysis example where it allows loading only a sub-set of runs. Instead, always all runs should be processed. This would then simplify fixing of the dataset: rename the existing files to match the expected split pattern, potentially load & save under a new name, where now MNE ensures that the split naming is done correctly.

larsoner, 2022-07-05:
IIRC the problem was that the raw file ended up being huge, so hosting it (and people reliably downloading it) were going to be tricky

drammock, 2022-07-05:
> the problem was that the raw file ended up being huge

if it's recreated *still as a split file* but with proper naming, this won't be an issue though right?

drammock, 2022-07-05:
it's 6ish GB...

[image]

If we fix this we should also get rid of the unnecessary "dotfiles".

My vote is for keeping them split, and fixing the naming so that they actually load as split files, as long as that doesn't have a huge impact on the tutorial run time.