mne-tools/mne-python #10914: .get_data() reject_by_annotation does not consider ch_names in annotations

FarnoodF, 2022-07-11:
#### Describe the new feature or enhancement
The .get_data() method in the mne.io.BaseRaw supports an argument called `reject_by_annotation` and the "bad" annotations can be either omitted or changed to `nan`. But as recently, a new feature was added to the `mne.Annotations` object to be able to add the `ch_names` for the annotations. The problem is that when the `ch_names` are set in the annotations, the `reject_by_channel` argument still omits or changes "all" the values to NaN.

#### Describe your proposed implementation
I think there should be an option to only transform those ch_names present in the annotations to NaN instead of all values.

agramfort, 2022-07-12:
indeed. Would you have time to make a PR about this?

> Message ID: ***@***.***>
>

FarnoodF, 2022-07-12:
@agramfort , sure, I can spend some time on it this week.