mne-tools/mne-python #10916: [ENH, MAINT] Check (and eventually test?) which API entries are not in examples

alexrockhill, 2022-07-12:
I feel like every couple weeks, I stumble upon a function that does something that I need that is already written in MNE and I wonder how I haven't come across it before. Examples are `mne.time_frequency.combine_tfr` and a lot of the early cluster-based stats stuff and associated functions that Dennis wrote. That makes me think that it would be nice to at least have a report of how many API entries are not used in any example. It could also help guide reorganizing examples and tutorials more sensibly if there were a connectivity plot of sorts of where the examples are used ideally built into the documentation. It could be required eventually that every new API entry have an example or `noqa` of sorts tagging it which I think is ideal. Very interested to know others' opinions. cc @drammock @agramfort @larsoner

drammock, 2022-07-13:
IMO there is a tension between what you suggest, and having examples/tutorials proliferate and not be systematically organized. Often the path of least resistance was taken --- to make a quick and dirty example with little or no narrative context --- instead of thinking about where in our existing tutorials it would make sense to integrate/ demo some new functionality. Having the list of un-demoed API elements seems good though... Just not sure I want to encourage any "quick fixes" to that list.

alexrockhill, 2022-07-13:
Agreed, I think it's less to add an example for the sake of an example and more of a road map plus now that the software is a bit more mature thinking about whether new API additions are in examples as they probably should be or be private.

larsoner, 2022-07-13:
This would be a nice enhancement to sphinx-gallery, as the information should be present in the "backreferences" to figure this out, and I think other projects would benefit. We could make a `:orphan:` HTML like https://mne.tools/dev/auto_tutorials/clinical/sg_execution_times.html that is an expert doc that gives this information, or just print a list of public functions + methods that are not backreferenced (probably subject to some `regex` matching so that some can be excluded, like our allow-list of explicitly public namespaces here: https://github.com/mne-tools/mne-python/blob/main/mne/tests/test_docstring_parameters.py#L20). @alexrockhill if you agree I'll transfer the issue over there.

larsoner, 2022-07-13:
I can't transfer to a new repo, so we should just make a new issue there. Can you do it @alexrockhill ?

alexrockhill, 2022-07-13:
Sure happy to, do you know if sphinx gallery has any Javascript plotting library dependencies yet? I can ask on the issue too

larsoner, 2022-07-13:
You mean for testing notebooks? These are not executed during SG generation

alexrockhill, 2022-07-13:
I mean for plotting a graph of API entries and examples.

larsoner, 2022-07-15:
For this I would maybe use `graphviz` or something, but really a list would be a good enough start.

alexrockhill, 2022-07-15:
Is `graphviz` a dependency of `sphinx_gallery`? I thought it was tricky to install also because it was only on `conda` and not `pip`. It's not a rush, definitely start with a list but it would be nice to know how one might go about doing that if/when it were to be added.

drammock, 2022-07-16:
graphviz is one of our doc build dependencies. You already have it installed, and so does CircleCI.

alexrockhill, 2022-07-16:
This is to go upstream in sphinx example gallery though. I don't think they have it as a dependency but that seems doable if it's on PyPI.

drammock, 2022-07-17:
Ah, right, I forgot this was going to go upstream. Anyway if they don't want to add the dependency it's still useful to have the list of un-demonstrated API entries

closed 2022-07-27 by alexrockhill