mne-tools/mne-python #10931: Confine the source space near the sensor

MA-Xiaoian, 2022-07-16:
Hello, I only have a few sensors, and I want to limit the source space to a certain distance from the sensor, similar to the function of mne .stc_ near_sensors, but it only for ECoG, EEG and DBS sensor data, how can I realize this function? Thanks ！

wmvanvliet, 2022-07-17:
There are [some functions](https://github.com/AaltoImagingLanguage/conpy/blob/master/conpy/forward.py) in [ConPy](https://users.aalto.fi/~vanvlm1/conpy/) that can do this. I do remember there was an effort at some point to put similar functionality in MNE-Python.

larsoner, 2022-07-25:
I think a clean way to do this would be to allow a user-supplied mask for the high-resolution mesh vertices to consider. Then people can arbitrarily mask points, which could include a suitable sensor-distance-based mask

wmvanvliet, 2022-07-26:
That is indeed the approach taken by conpy.