mne-tools/mne-python #10932: Version menu is getting too long

cbrnr, 2022-07-17:
Somewhat related to #10789, our version menu is getting pretty long (and if we don't do anything about it, it will get too long rather soon):
[image]

I propose that in addition to *devel* and *stable*, we only show *five* previous versions and then maybe a link to a page with older versions. Alternatively, we might just drop older versions altogether – or is there a particular reason why someone would want to use the documentation for MNE-Python that's over 3 years old?

larsoner, 2022-07-17:
Another option might be a little CSS fix like
```
.whatever_drop_down
{
   height: 50px;
   overflow: auto;
}
```
I'm also okay with making the list shorter. Happy to defer to @drammock on this

larsoner, 2022-07-17:
(the above or some similar variant would add a scroll bar to the dropdown)

cbrnr, 2022-07-17:
A scrollbar would also work, but at some point it too will get unwieldy. I'm fine with a short-term and medium-term solution though.

PR #11888 (2023-08-16): MAINT: Clean up whats_new and doc versions
- Closes #11885
- Closes #10932
- Closes #10789 

1. Refactor `whats_new` to show latest changes and link to all others
2. Refactor `rst_prolog` so we don't need to `.. include:: links.inc` and `.. currentmodule:: mne` all over the place
3. Add links to old doc version archives
4. Streamline `links.inc` to be much shorter and remove a lot of stuff we don't use (didn't check everything but checked some stuff at least)

@drammock let me know what you think. @sappelhoff also feel free to look since you had a preference for the long, continuous `whats_new [... cut]
closed 2023-08-16 by larsoner