mne-tools/mne-python #10934: Adding the ability to only display a subset of the ICA components in "plot_sources"

adam2392, 2022-07-19:
#### Describe the new feature or enhancement
In mne-icalabel GUI (PR here https://github.com/mne-tools/mne-icalabel/pull/66), we would like to plot one ICA source at a time. We are using MNE-QT-Browser to embed a time-series plot of the ICA source one at a time. We are using the LOC suggested in https://github.com/mne-tools/mne-qt-browser/issues/117.

Currently `ica.plot_sources()` doesn't allow you to plot a subset of sources, instead it shows all of them at once. I would like to only plot one at a time depending on which ICA component is selected.

#### Describe your proposed implementation
In https://mne.tools/stable/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA.plot_sources, we would add the kwarg `ic_picks=None`, which allows one to pick a subset of the ICA sources to plot.

mscheltienne, 2022-07-19:
Actually it's already possible. The `picks` argument does not refer to the instance channels but to the ICs. 

Example:

```
import os
import mne
from mne.preprocessing import ICA

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
# Here we'll crop to 60 seconds and drop gradiometer channels for speed
raw.crop(tmax=60.).pick_types(meg='mag', eeg=True, stim=True, eog=True)
raw.load_data()

ica = ICA(n_components=5, method="picard")
ica.fit(raw, picks="eeg")
```

If you do `ica.plot_sources(raw, picks=[0, 1])` you get the sources for IC000 and IC001; and if you do `ica.plot_sources(raw, picks=[0, 1, 2, 3, 4, 5]` you get the following error as only 5 components have been fitted:

```
ValueError: All picks must be < n_channels (5), got [0, 1, 2, 3, 4, 5]
```

The docstring(s) and the error(s) should be updated to refer to the ICs instead of the channels. It's not the only ICA-related method with a `picks` argument that refers to the ICs.

drammock, 2022-07-19:
I've updated the issue title to reflect that the problem here is in the docstring and the error message.

PR #10936 (2022-07-19): [DOC] Fix ICA plot_sources "picks" documentation
#### Reference issue
Fixes: #10934 

#### What does this implement/fix?
Fix the doc string.
closed 2022-07-19 by drammock