mne-tools/mne-python #10937: [BUG] Hilbert transform returns negative data with `envelope=True`

alexrockhill, 2022-07-19:
Minimally reproducible example:

```
import numpy as np
import mne
from mne.datasets import sample

data_path = sample.data_path()
subjects_dir = data_path / 'subjects'
raw_fname = data_path / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(raw_fname)
raw.info['bads'] = ['MEG 2443']  # bad MEG channel
event_id = 1  # those are the trials with left-ear auditory stimuli
tmin, tmax = -0.2, 0.5
events = mne.find_events(raw)
raw.pick(['meg', 'eeg', 'eog'])
epochs = mne.Epochs(raw, events, event_id, tmin - 0.3, tmax + 0.3,
                    baseline=(None, 0), preload=True, proj=False,
                    reject=dict(grad=4000e-13, mag=4e-12,
                                eeg=1e-4, eog=150e-6))
epochs = epochs[:10]
epochs.set_eeg_reference('average')

bandwidth = 2
epochs.filter(l_freq=12 - bandwidth / 2, h_freq=12 + bandwidth / 2)
epochs.apply_hilbert(envelope=True)
print(epochs.get_data().min())
```

I thought this was not supposed to be negative, I'm not sure what it means that it is negative.

agramfort, 2022-07-20:
hilbert does not apply to EOG. The negative value is on the EOG channel

> Message ID: ***@***.***>
>

alexrockhill, 2022-07-20:
Ah really should have noticed that!

closed 2022-07-20 by alexrockhill