mne-tools/mne-python #10941: Duplicate functin check:check_version and _testing:check_version with 2 different behaviors

mscheltienne, 2022-07-21:
I was trying to figure out why running with `main` (1.1.0.dev):

```
from mne.utils import check_version, requires_version

check_version("mne", "1.1")  # returns True
requires_version("mne", "1.1")  # marks to skip, i.e. check_version returned False
```

And it turns out there is both https://github.com/mne-tools/mne-python/blob/c6a86f15b8ace3758443f392ba8bfe5a9a0c2878/mne/utils/check.py#L69-L120

and https://github.com/mne-tools/mne-python/blob/c6a86f15b8ace3758443f392ba8bfe5a9a0c2878/mne/utils/_testing.py#L170-L197

Which don't exactly do the same thing, but maybe they should?

larsoner, 2022-07-25:
Yes the second one should behave like the first where `1.1.dev0` is acceptable when requiring `1.1`. A PR to fix this would be great!

mscheltienne, 2022-07-25:
Thanks for the confirmation, PR opened, 🤞 for green CIs.

PR #10958 (2022-07-26): MRG: Replace check_version from _testing.py with the same function in check.py
Closes #10941
closed 2022-07-26 by larsoner