mne-tools/mne-python #10948: TIME UNIT IN EVOKE PLOT

isabelceo, 2022-07-22:
I suggets that in Evoked.plot(), plot_compare_evokeds() has time_unit (ms) because it is easier for doctor and neurologists to interpret the resuls.
I hope you have this consideration

agramfort, 2022-07-22:
you mean add time_unit param to plot_compare_evokeds?

evoked.plot(time_unit='ms') should already does what you want?

hoechenberger, 2022-07-22:
> you mean add time_unit param to plot_compare_evokeds?

Yes, that's what's being requested here

isabelceo, 2022-07-22:
Yes, that's I suggets, add time_unit param

drammock, 2022-07-22:
`Evoked.plot()` *already has* a `time_unit` parameter, since version 0.16. Scroll down a bit you'll see it's highlighted yellow:

https://mne.tools/stable/generated/mne.Evoked.html?highlight=time_unit#mne.Evoked.plot

drammock, 2022-07-22:
oh, sorry, I misunderstood. You want `time_unit` added to `plot_compare_evokeds`.  Sure, I can do that.

PR #10950 (2022-07-22): add time_unit to plot_compare_evokeds
closes #10948
closed 2022-07-22 by drammock