mne-tools/mne-python #10952: ImportError: cannot import name 'tqdm' from 'mne.externals'

xiaohao-lin1, 2022-07-24:
#### Describe the bug
ImportError: cannot import name 'tqdm' from 'mne.externals' (D:\Users\SCSE-AI.R-CIL1\anaconda3\envs\MoBI_py3.8\lib\site-packages\mne\externals\__init__.py)

#### Steps to reproduce
I was using the autoreject package, which calls upon the MNE package, and encountered this error. 
from autoreject import AutoReject
ar = AutoReject()
epochs_clean = ar.fit_transform(epochs)  

  File "D:\Users\SCSE-AI.R-CIL1\anaconda3\envs\MoBI_py3.8\lib\site-packages\mne\utils\progressbar.py", line 49, in __init__
    import tqdm
ImportError: cannot import name 'tqdm' from 'mne.externals' (D:\Users\SCSE-AI.R-CIL1\anaconda3\envs\MoBI_py3.8\lib\site-packages\mne\externals\__init__.py)

#### Expected results
*Replace this text with a description of what you expected to happen.*

#### Actual results
*Replace this text with the actual output, traceback, screenshot, or other
description of the results.*

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*

cbrnr, 2022-07-24:
Can you post the output of `mne.sys_info()` please? It seems like you are using an old version, because `tqdm` has been removed from `mne.externals`.

agramfort, 2022-07-24:
You need to update the autoreject package 

This issue has been fixed a few months ago

xiaohao-lin1, 2022-07-24:
> Can you post the output of `mne.sys_info()` please? It seems like you are using an old version, because `tqdm` has been removed from `mne.externals`.

Hi I have updated mne.sys_info() in the post

xiaohao-lin1, 2022-07-24:
> You need to update the autoreject package This issue has been fixed a few months ago

Hi I am currently using autoreject version 0.4.dev0.

I have tried using the stable release of 0.3.1 too, but I encountered the same issue. I am using mne mne: 0.24.0

cbrnr, 2022-07-24:
Can you upgrade `mne`? The latest version is 1.0.3, and this includes `tqdm` as a dependency.

larsoner, 2022-09-15:
I think this has been fixed and is an old MNE/autoreject issue

closed 2022-09-15 by larsoner