mne-tools/mne-python #10959: mne.time_frequency.psd_welch with option average=median

chapochn, 2022-07-25:
In the function `mne.time_frequency.psd_welch`, for the option `average` it is written:
_How to average the segments. If mean (default), calculate the arithmetic mean. If median, calculate the median, corrected for its bias relative to the mean. If None, returns the unaggregated segments._

I searched through the code, and am not finding any correction of the median for the bias relative to mean.
1. Is it actually done somewhere and I missed?
2. Would you have some references about how such bias correction is done?

larsoner, 2022-07-26:
I'm not aware of a problem with PSD bias when using median vs mean (but there could be one!). Do you have refs for this?

chapochn, 2022-07-26:
I googled but didn't find so far. I had no awareness that there needed to be a bias correction. But am interested in learning about it!

agramfort, 2022-07-27:
I have no idea where this comes from...

drammock, 2022-07-27:
it was added here: https://github.com/mne-tools/mne-python/commit/88a2e3dae4b22ca13d088db8b2213745cd9a653c

@hoechenberger authored and @agramfort committed :)

agramfort, 2022-07-27:
@hoechenberger what did you mean by "corrected for its bias relative to the
mean" ?

> Message ID: ***@***.***>
>

hoechenberger, 2022-07-27:
No idea, it's probably just rephrasing what the invoked SciPy function actually does. IIRC it's not simply calculating a median but doing some magic to avoid a certain type of bias. But I'm not on my computer now, so cannot really look. But the solution lies in the SciPy docs & code.

drammock, 2022-07-27:
maybe from here?

> If we now introduce a discontinuity in the signal, by increasing the amplitude of a small portion of the signal by 50, we can see the corruption of the mean average power spectral density, but using a median average better estimates the normal behaviour.

https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.welch.html

(near the end of the page)

hoechenberger, 2022-07-27:
https://pydocs.github.io/p/scipy/1.8.0/api/scipy.signal._spectral_py._median_bias

It's this function that's called internally.

mscheltienne, 2022-07-27:
Are you sure? Following the `average` trail leads to https://github.com/mne-tools/mne-python/blob/c12a02000776adb9df18da55ea60c50cbf83827d/mne/time_frequency/psd.py#L20-L21

hoechenberger, 2022-07-27:
Well maybe somebody broke things. 🤓

Back when I added this stuff (follow the link to the commit @drammock dug up) we used some SciPy functions that would apply this bias correction. 

Basically the calculated median would get divided by that bias to retrieve the corrected median.

mmagnuski, 2022-08-01:
I may have been involved in the `np.nanmedian` because previous approach did not work in some cases (when some welch windows were NaN due to overlap with annotations, I think), so I submitted an issue, which lead to nanmedian. It would be great to have both then: the median bias correction, and ability to ignore nans.

mmagnuski, 2022-08-01:
([here is the scipy function BTW](https://github.com/scipy/scipy/blob/main/scipy/signal/_spectral_py.py#L2019))

hoechenberger, 2022-08-01:
We could just steal that bias estimation function from SciPy, it's just two lines of code…

agramfort, 2022-08-02:
we should fix this before the release

> Message ID: ***@***.***>
>

PR #10990 (2022-08-02): BUG: Fix bug with median bias
Closes #10959
closed 2022-08-02 by drammock