mne-tools/mne-python #10964: Subject weight info key is not documented

hoechenberger, 2022-07-27:
We support `info['subject_info']['weight']`:
https://github.com/mne-tools/mne-python/blob/c12a02000776adb9df18da55ea60c50cbf83827d/mne/io/meas_info.py#L1694

but the `weight` key is not documented at:
https://mne.tools/dev/generated/mne.Info.html

sappelhoff, 2022-07-27:
What does it even do? Is it "1" in normal cases and can be set to more or less than that when averaging wicth other subjects?

hoechenberger, 2022-07-27:
I don't think I understand your question … it's just the weight in an unspecified unit, e.g., in kg or lb, stored as a float.

hoechenberger, 2022-07-27:
Just figured that height isn't documented either.

https://github.com/mne-tools/mne-python/blob/c12a02000776adb9df18da55ea60c50cbf83827d/mne/io/meas_info.py#L1697

sappelhoff, 2022-07-27:
>  it's just the body weight 

oh, ... okay.

> in an unspecified unit

O.o

larsoner, 2022-07-27:
Units are kg and m

https://github.com/mne-tools/fiff-constants/blob/30d89d420d50c9e465f46524d4901795dac66741/DictionaryTags.txt#L378-L379

sena-neuro, 2022-08-09:
Hello! May I take this as my first issue?

agramfort, 2022-08-09:
Sure ! 🙏

PR #11019 (2022-08-10): [DOC, MRG] weight & height keys of subject_info dict is documented
Closes #10964.

`height` and `weight` keys of  `subject_info` entry in `mne.Info` are documented.
closed 2022-08-10 by drammock