mne-tools/mne-python #10965: DOC: Document inst better

larsoner, 2022-07-27:
In most Returns we just do `instance of Raw, Epochs, or Evoked`. It's not optimal and in some cases not even complete (e.g., can be AverageTFR, STC, etc. instead) so we could improve this.

Options:

1. Create a new category like `MNE-object` with a glossary entry
2. Say something like `instance of original type` or something more compact that has this same meaning to say "you get back the same instance mostly for chaining purposes", maybe with a glossary entry.
3. Figure out some magical way to put the class name in the super class doc. This isn't possible via autodoc, so we'd maybe have to do some runtime replacement. For example if we decorated each class with a `@copy_name_to_methods` we could replace every method with one that has the same signature, wraps directly to the super, but replaces `__doc__` with one filled with the appropriate class name. But this is a lot of work for little gain.

I actually prefer (2) to (3) even if we can figure out a way to do (3) because it makes it clear you're actually getting back the original object for chaining purposes. In cases where it is a copy / different instance, we should explicitly say so somehow probably, maybe with another name like `copied instance of same type` or something. But we'd need to think about how to do this in a NumpyDoc-acceptable manner.

_Originally posted by @larsoner in https://github.com/mne-tools/mne-python/pull/10945#discussion_r930217055_

hoechenberger, 2022-08-01:
When read the title of this issue, I was actually expecting (hoping) it to be about all the places where we emit log messages (or have parts in the documentation) that literally mention "inst", e.g. something like, "… call inst.some_method()" or so (I don't have an explicit example now)
I think these are downright confusing for "common" users, esp. if nowhere in the signature of the function that produced this message an `inst` parameter exists.

But yeah … that's another issue, it seems

cbrnr, 2022-08-01:
👍 for (2). There is a new `Self` type in `typing`, but let's better not go down this route.

hoechenberger, 2022-08-01:
> but let's better not go down this route.

😏

cbrnr, 2022-08-01:
> I have to say I never understood why we are explicitly prepending "instance of". It's inconsistent with how we specify input types, where we usually don't have these two words.

Absolutely agree! Especially in Python, where everything is an object (aka instance). I'd remove this.

hoechenberger, 2022-08-01:
> Absolutely agree! Especially in Python, where everything is an object (aka instance). I'd remove this.

I assume the thinking was something like, "If the returned object is `mne.Epochs`, what we actually return is not a pointer to `mne.Epochs`, but to the result of `mne.Epochs(...)`, i.e., an instance". But … this is **so** trivial that it's downright confusing because it requires thinking where usually one doesn't need to think, at least not if you're a user 😀

drammock, 2022-08-02:
> I actually prefer (2) to (3) even if we can figure out a way to do (3) because it makes it clear you're actually getting back the original object for chaining purposes. In cases where it is a copy / [...]

I agree that (2) is the best option, though IMO it is pretty tricky to come up with simple, unambiguous wording. Here's an (unsatisfactory) attempt:

```
Returns
-------
inst : self
    An instance of the object type whose method was called. If ``copy=True``
    this will be a copy with <WHATEVER THE METHOD CHANGED>, otherwise the
    original instance will be modified in-place and returned.
```

cbrnr, 2022-08-02:
What about:

```
Returns
-------
inst : self
    The modified object if ``copy=False``. If ``copy=True``,
    this will be a copy with <WHATEVER THE METHOD CHANGED>.
```

I'd also drop the name of the return value if there is only one, i.e.

```
Returns
-------
self
    The modified object if ``copy=False``. If ``copy=True``,
    this will be a copy with <WHATEVER THE METHOD CHANGED>.
```

I'm not 100% happy with this either.

hoechenberger, 2022-08-02:
> Returns
> -------
> self

I don't think that's very helpful to non-developers.

Why can we not just spell it out?

```
Returns
-------
Raw, Epochs, or Evoked – same as input type
```
or so?

drammock, 2022-08-02:
> and be explicit for a method and only list the one type that it applies to

because for methods that come from mixins, we don't know the type at the time the doc is being written. Hence the mention of "the super class doc" in the OP.

hoechenberger, 2022-08-02:
> > and be explicit for a method and only list the one type that it applies to
> 
> because for methods that come from mixins, we don't know the type at the time the doc is being written. Hence the mention of "the super class doc" in the OP.

Than how about the poor person's approach of

```
Returns
-------
result : same type as the input data
```
?

drammock, 2022-08-02:
> result : same type as the input data

that's not a bad proposal in terms of clarity, but whatever we do has to pass `numpydoc` and/or be special-cased so that it passes `numpydoc`.  Therefore fewer words are better, and I'm a bit hesitant to special-case words like `input` and `data`.

hoechenberger, 2022-08-02:
> > result : same type as the input data
> 
> that's not a bad proposal in terms of clarity, but whatever we do has to pass `numpydoc` and/or be special-cased so that it passes `numpydoc`. Therefore fewer words are better, and I'm a bit hesitant to special-case words like `input` and `data`.

How about

```
result : same type as inst
```
?

larsoner, 2022-09-15:
@hoechenberger I see you milestoned this for 1.2 but we converge and have a volunteer to work on this. I think it's not likely to happen in the next ~week

hoechenberger, 2022-09-15:
No worries, let's bump this to 1.3 then!

larsoner, 2024-07-12:
Milestone on this has been bumped numerous times, removing milestone until someone champions this issue. But also marking as "easy" so it hopefully gets visibility / higher likelihood of being tackled

larsoner, 2025-11-26:
@keerthanam0905-22 you can just open a PR when you're ready, no need to assign. I guess we decided this was the best pattern:
```
Returns
-------
result : same type as the input data
```
Maybe start by changing one instance of this and we can make sure we're happy with how it looks (and make `numpydoc` tolerate it), then you could change other instances.