mne-tools/mne-python #10976: Plot function for virtual channels in bridging electrode EEG analysis

adam2392, 2022-07-29:
#### Describe the new feature or enhancement
The nice tutorial made by @alexrockhill in https://mne.tools/dev/auto_examples/preprocessing/eeg_bridging.html#sphx-glr-auto-examples-preprocessing-eeg-bridging-py shows how one can plot the "virtual" `<bridge_elec1>-<bridge_elec2>` signal and visually confirm if the data is in fact bridged.

I would like to have this plot as a function to enable someone's workflow.

#### Describe your proposed implementation
The https://mne.tools/dev/generated/mne.preprocessing.interpolate_bridged_electrodes.html#mne.preprocessing.interpolate_bridged_electrodes function internally has a method for computing the "virtual channels" for the sake of interpolation. 

I would just refactor that to a private method, and then add a plot method for `plot_bridged_virtual_ch` to recreate this plot but for all bridged electrode pairs: https://mne.tools/dev/auto_examples/preprocessing/eeg_bridging.html#plot-the-raw-voltage-time-series-for-bridged-electrodes

#### Additional comments
I don't think we want to return the full Raw object probably(?), but the plot is generated from a "Raw" copy of the original data + `bridged_idx`. Should we plot all virtual channels? This might be cluttered because there could be MANY virtual channels...?

WDYT? @alexrockhill

alexrockhill, 2022-07-29:
Awesome, I'm all for it! It is possible that people would want to continue on with the virtual channels rather than using them to interpolate so that would be nice to support if I'm understanding you correctly. Also, if it were in a method (maybe in a mixin) that would be a great improvement IMO. Thanks for iterating on this!

adam2392, 2022-07-29:
> Awesome, I'm all for it! It is possible that people would want to continue on with the virtual channels rather than using them to interpolate so that would be nice to support if I'm understanding you correctly. 

Oh great point! Then prolly better to return a Raw object that one can then plot on their own...

Would someone prefer to use the virtual channels or the interpolation? Is there a recommendation?

Also unrelated, how does pyprep/auto reject fit into this? Those mainly help automate what's a bad channel on the raw time series? So maybe if someone wants to not reject too much data, they stick with bridging, filtering and (automated) ica analysis? Otw they can use pyprep/auto reject with bridging?

alexrockhill, 2022-07-29:
After a second reading, I think I totally misread that, you're talking about a n_channels x n_channels - 1 raw plot. In teaching undergrads EEG, they asked for this too so I think people want to see this info but it is tricky with the sheer number of channels. Maybe a non-scrolly plot but more like an evoked plot would be helpful to be able to see everything. Then you wouldn't get an exact ID on the bridged channels (until running the algorithm) but you could get a sense of your data. I'm really not sure, it seems to be a practical question. With so many EEG channels, the issue for sure is dimensionality, whether that's limiting the number of virtual channels per scroll page or some more out of the box solution like above, I think that's the tricky part. There's my 2 cents but I'm not sure I can save you the effort of mocking it up and seeing if the version you think is best manageable, it's just a lot of data!