mne-tools/mne-python #10977: Update theme pin and check docs build

drammock, 2022-07-29:
**About a week before a new release**, make a PR that updates our pin of `pydata-sphinx-theme` to the theme's current `HEAD` commit, and do a full docs build (`[circle full]`).  If anything looks awry, do some combination of 

1. working backwards on the theme's tree until you find a commit that works/looks good on our site
2. reporting bugs to the theme (and ideally fixing them and updating the commit hash accordingly)

When you're satisifed that our dev site looks good, **DO NOT CLOSE THIS ISSUE!**  Instead, just bump its milestone.

Places to update the pin:

- `requirements_doc.txt`
- sometimes in the past we've done something different in `tools/circleci_dependencies.sh` so double-check if a change is needed there

drammock, 2022-08-01:
No I agree that this was dealt with for 1.1 by #10973

larsoner, 2023-01-16:
@drammock how feasible would it be to pin to a released version? It looks like ~4 minutes is spent in every SphinxWindows Azure build to `pyproject.toml` build the pinned commit hash, which we should be able to skip if we use a wheel on PyPI

drammock, 2023-01-16:
Should be ok. A new release is due in a week or two, if that matters.

PR #11557 (2023-03-15): MAINT: Bump minimums
Bumps min reqs in README.rst to those ~2 years old this April, and adjusts/removes code shims for deps lower than those. Also `nibabel` is now required for `read_surface` which I think is reasonable (it is pure Python and has been stable and pip/conda-installable for years now).

Closes #10977
closed 2023-03-15 by larsoner