mne-tools/mne-python #10984: One test failure on aarch64

musicinmybrain, 2022-08-01:
#### Describe the bug

While packaging version 1.0.3 for Fedora linux, we found that the test `test_mxne_inverse_sure_synthetic[3-0.75-4-20-60-20]` fails on `aarch64` due to division by zero.

#### Steps to reproduce

Run the unit tests on an `aarch64` system. There is no reason to believe this error is specific to the Fedora Linux RPM build environment.

#### Expected results

All tests pass.

#### Actual results

```
=================================== FAILURES ===================================
_____________ test_mxne_inverse_sure_synthetic[3-0.75-4-20-60-20] ______________
mne/inverse_sparse/tests/test_mxne_inverse.py:372: in test_mxne_inverse_sure_synthetic
    _, active_set, _ = _compute_mxne_sure(M, G, alpha_grid, sigma=sigma,
<decorator-gen-507>:10: in _compute_mxne_sure
    ???
mne/inverse_sparse/mxne_inverse.py:881: in _compute_mxne_sure
    coefs_grid_1, coefs_grid_2, active_sets = _fit_on_grid(gain, M, eps, delta)
mne/inverse_sparse/mxne_inverse.py:858: in _fit_on_grid
    X_eps, a_set_eps = _run_solver(alpha, M_eps,
mne/inverse_sparse/mxne_inverse.py:819: in _run_solver
    X, active_set, _ = iterative_mixed_norm_solver(
<decorator-gen-497>:10: in iterative_mixed_norm_solver
    ???
mne/inverse_sparse/mxne_optim.py:584: in iterative_mixed_norm_solver
    X, _active_set, _ = mixed_norm_solver(
<decorator-gen-496>:10: in mixed_norm_solver
    ???
mne/inverse_sparse/mxne_optim.py:485: in mixed_norm_solver
    X, active_set, E = l21_solver(M, G, alpha, lc, maxit=maxit,
<decorator-gen-495>:12: in _mixed_norm_solver_bcd
    ???
mne/inverse_sparse/mxne_optim.py:233: in _mixed_norm_solver_bcd
    c = z / z.sum()
E   RuntimeWarning: divide by zero encountered in true_divide
```

#### Additional information

```
Platform:         Linux-5.18.11-200.fc36.aarch64-aarch64-with-glibc2.35.9000
Python:           3.11.0b5 (main, Jul 26 2022, 00:00:00) [GCC 12.1.1 20220628 (Red Hat 12.1.1-3)]
Executable:       /usr/bin/python3
CPU:              aarch64: 5 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.0.3
numpy:            1.22.0 {blas=flexiblas, lapack=flexiblas}
scipy:            1.8.1
matplotlib:       3.5.2 {backend=agg}
sklearn:          1.0.2
numba:            Not found
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.3.5
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              9.1.0
PyQt5:            5.15.6
ipympl:           Not found
pooch:            v1.5.2
mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
```

I am happy to run any additional tests that may be helpful, or try any candidate fixes, on real or emulated `aarch64` hardware.

musicinmybrain, 2022-08-01:
@sanjayankur31

larsoner, 2022-10-27:
I think this should be fixed by https://github.com/mne-tools/mne-python/pull/11284

closed 2022-10-28 by larsoner

musicinmybrain, 2022-10-30:
> I think this should be fixed by #11284

I [tried applying that PR](https://src.fedoraproject.org/rpms/python-mne/pull-request/9) as a patch in the Fedora Linux package, and I no longer see failures on aarch64 or on ppc64le (#10985)—so I think it worked!

PR #11284 (2022-10-28): BUG: Work around ppc64le bugs
Closes #10985
Closes #10984

Ubuntu 22.10 fixed the QEMU bug that was stopping me from tracking this down. There must be a bug in OpenBLAS 3.19 with `np.convolve`->`multiarray.correlate` for complex arrays, but I haven't followed the chain to see what this wraps to under the hood. In the meantime, this workaround should be fine and have a negligible performance hit.