mne-tools/mne-python #10991: Surfaces and sensors not showing in coregistration/alignment visualization

olafhauk, 2022-08-02:
#### Describe the bug
*Replace this text with a description of the bug.*

Using mne.viz.plot_alignment() (as e.g. in tutorial 20_source_alignment.html only shows landmarks but not surfaces or sensors.
Version 1.0.3 under Linux (Centos 7).

#### Steps to reproduce

tutorial 20_source_alignment.html

#### Expected results

Plots as in tutorial https://mne.tools/stable/auto_tutorials/forward/20_source_alignment.html.

#### Actual results

[image]

#### Additional information

Platform:         Linux-3.10.0-1160.el7.x86_64-x86_64-with-glibc2.17
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:38:57) [GCC 10.3.0]
Executable:       /imaging/local/software/mne_python/mne1.0.3_0/bin/python
CPU:              x86_64: 20 cores
Memory:           251.7 GB

mne:              1.0.3
numpy:            1.21.6 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.2 {backend=QtAgg}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.34.0 {OpenGL 3.3 (Core Profile) Mesa 18.3.4 via llvmpipe (LLVM 7.0, 256 bits)}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
PyQt5:            5.12.3
ipympl:           0.9.1
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         0.2.1
mne_features:     0.2
mne_qt_browser:   0.3.1
mne_connectivity: 0.3

closed 2022-10-21 by larsoner

mscheltienne, 2024-05-13:
I'm getting the same behavior on a documentation build on github action, but not running old versions of MESA. Relevant part of a `sys_info`:

```
Platform:                 Linux-6.5.0-1018-azure-x86_64-with-glibc2.35
Python:                   3.9.19 (main, Mar 20 2024, 16:40:02)  [GCC 11.4.0]
Executable:               /opt/hostedtoolcache/Python/3.9.19/x64/bin/python
CPU:                      x86_64
Physical cores:           2
Logical cores:            4
RAM:                      15.6 GB
SWAP:                     4.0 GB

Core dependencies
✔︎ mne (>=1.7):            1.7.0
✔︎ numpy (>=1.21):         1.26.4

Optional 'doc' dependencies
✔︎ matplotlib:             3.8.4 (backend: QtAgg)
✔︎ PyQt6:                  6.7.0
✔︎ pyvista:                0.43.7 (OpenGL 4.5 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2 via llvmpipe (LLVM 15.0.7, 256 bits))
✔︎ pyvistaqt:              0.11.0
✔︎ sphinx (!=7.2.*):       7.3.7
✔︎ sphinx-gallery:         0.16.0
```

The associated [GA action](https://github.com/fcbg-hnp-meeg/meg-wiki/blob/main/.github/workflows/doc.yml) setups a headless display with `pyvista/setup-headless-display-action@main`.
I'm going to try to change the runner, reporting in case you have an idea to spare me some headaches :exploding_head:

larsoner, 2024-05-13:
Can you see if any of these help?

https://mne.tools/dev/install/advanced.html#off-screen-rendering-with-mesa

In particular something like
```
export MNE_3D_OPTION_MULTI_SAMPLES=1
export MNE_3D_OPTION_ANTIALIAS=false
```
might fix it

mscheltienne, 2024-05-13:
Thank you @larsoner, we have a winner `LIBGL_ALWAYS_SOFTWARE: true` does the trick.
I'm not very familiar with 3D rendering, but should this variable be part of the variables defined by `pyvista/setup-headless-display-action@main`, namely `DISPLAY: :99.0` and `PYVISTA_OFF_SCREEN: true`? Or is there a downside for some workflows in settings this environment variable on github runners?

mscheltienne, 2024-05-13:
Actually `MNE_3D_OPTION_MULTI_SAMPLES: 1` is also needed, I probably got baited by some cache on my browser.

PR #11271 (2022-10-21): BUG: Correctly detect MESA
Closes #10991

Fixes a bug with Mesa detection, and now emits a warning if someone is using a version of MESA that is too old to work.