mne-tools/mne-python #10992: mne.gui.coregistration() helmet and surfaces not shown correctly

olafhauk, 2022-08-02:
#### Describe the bug

mne.gui.coregistration() shows scalp surface and digitization points, but:
1) not the MEG helmet
2) responds to "Head opacity" in "all-or-none" manner, i.e. scalp surface is only shown when head opacity slider is at its rightmost position.
Version 1.0.3 (Linux, Centos 7)

#### Steps to reproduce
mne.gui.coregistration(subject='sample', subjects_dir='...MNE-sample-data/subjects', inst='.../MEG/sample/sample_audvis_raw.fif')

#### Expected results

Plots e.g. in https://mne.tools/stable/auto_tutorials/forward/25_automated_coreg.html.

#### Actual results
With scalp surface, no MEG helmet:
[image]

No scalp surface, no MEG helmet:

[image]

#### Additional information

Platform:         Linux-3.10.0-1160.el7.x86_64-x86_64-with-glibc2.17
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:38:57) [GCC 10.3.0]
Executable:       /imaging/local/software/mne_python/mne1.0.3_0/bin/python
CPU:              x86_64: 20 cores
Memory:           251.7 GB

mne:              1.0.3
numpy:            1.21.6 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.2 {backend=QtAgg}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.34.0 {OpenGL 3.3 (Core Profile) Mesa 18.3.4 via llvmpipe (LLVM 7.0, 256 bits)}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
PyQt5:            5.12.3
ipympl:           0.9.1
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         0.2.1
mne_features:     0.2
mne_qt_browser:   0.3.1
mne_connectivity: 0.3

larsoner, 2022-08-02:
I suspect that this an https://github.com/mne-tools/mne-python/issues/10991 are both to translucent geometry not working on VTK+Mesa 18.3.6.

I assume you can't upgrade to CentOS 8 or something else with a newer MESA...?

larsoner, 2022-09-15:
From debugging with @olafhauk we determined that this is an issue with MESA 18.3.4 and there isn't much we can do there but wait for ancient machines to be upgraded. Old CentOS is close to EOL anyway so hopefully soon it's a non-issue. In the meantime, our suggestions for how to deal with old MESA already in our docs are good enough I think.

closed 2022-09-15 by larsoner