mne-tools/mne-python #10996: I don't think the bias calculation is correct in the case where there are nans in the `spect`. I believe `_median_bias` needs to take the number of non nans in its argument, no?

drammock, 2022-08-02:
I don't think the median bias calculation is correct in the case where there are nans in the `spect`. I believe `_median_bias` needs to take the number of non nans in its argument, no?

_Originally posted by @chapochn in https://github.com/mne-tools/mne-python/pull/10990#discussion_r935753477_

mmagnuski, 2022-08-02:
I'm on it now.

larsoner, 2022-08-02:
I wasn't sure if you were still working on it so I at least started thinking about it, this is close but not there:
```
# adapted from SciPy
# https://github.com/scipy/scipy/blob/f71e7fad717801c4476312fe1e23f2dfbb4c9d7f/scipy/signal/_spectral_py.py#L2019  # noqa: E501
def _median_biases(n):
    # Compute the biases for 0 to max(n, 1) terms included in a median calc
    biases = np.ones(n + 1)
    # The original SciPy code is:
    #
    # def _median_bias(n):
    #     ii_2 = 2 * np.arange(1., (n - 1) // 2 + 1)
    #     return 1 + np.sum(1. / (ii_2 + 1) - 1. / ii_2)
    #
    # This is a sum over (n-1)//2+1 terms.
    # Terms here for different n are:
    # 0 = []  # 0 terms
    # 1 = []  # 0 terms
    # 2 = [1]  # 1 term
    # 3 = [1]  # 1 term
    # 4 = [1, 2]  # 2 terms
    # 5 = [1, 2]  # 2 terms
    # ...
    # We can get the terms for 0 through n using a cumulative summation and
    # indexing:
    if n > 1:
        ii_2 = 2 * np.arange(1, (n - 1) // 2 + 1)
        sums = 1 + np.cumsum(1. / (ii_2 + 1) - 1. / ii_2)
        biases[2:] = sums[np.arange(n - 1) // 2]
    return biases
```

larsoner, 2022-08-02:
... then I was going to get the biases with something like:
```
    elif average == 'median':
        biases = _median_biases(spect.shape[-1])
        idx = (~np.isnan(spect)).sum(-1)
        spect = np.nanmedian(spect, axis=-1) / biases[idx]
```

larsoner, 2022-08-02:
(I think it's useful to keep this vectorized, otherwise we're going to get hurt by the fact that these can be n_epochs x n_freqs in shape, i.e., could be tens or hundreds of thousands of calls)

mmagnuski, 2022-08-02:
Take a look at the PR, I first wanted to make this vectorized in a way that the number of nans can be different, but then realized it should be the same for every epoch / frequency so it does not make sense to be overly universa.

mmagnuski, 2022-08-02:
but you can modify my PR if you prefer the more universalized version :)

mmagnuski, 2022-08-02:
well, I didn't actually think about epochs - the number of nans should be the same for every frequency, but not for every epoch. So indeed the fully vectorized approach should be better.

drammock, 2022-08-03:
closed by #10997

closed 2022-08-03 by drammock