mne-tools/mne-python #11005: Reading EEGLAB requires `pymatreader`

cbrnr, 2022-08-04:
Reading EEGLAB files requires `pymatreader`, which however is not a hard dependency. This makes it more difficult for other packages that depend on MNE to load such files, because they have to explicitly list `pymatreader` as a dependency.

In addition, most EEGLAB files I have seen do not use the v7.3 (HDF5) format, so they could be read with `scipy.io.loadmat()`, which is already a hard dependency.

Therefore, I see two options here:

1. Make `pymatreader` a hard dependency.
2. Fall back to `scipy.io.loadmat()` if `pymatreader` is not installed.

Option 2 implies that when trying to read v7.3 files with the fallback, users would get an error (after which they could still decide to install `pymatreader`). Furthermore, some tests need to be adapted to distinguish between v7.3 and <v7.3 files (currently 4 out of 17 tests fail).

cbrnr, 2022-08-04:
See also https://eeglab.org/tutorials/03_Dataset_management/datasets.html#dataset-and-file-preferences, where they recommend storing to <v7.3 and only use v7.3 if you have files larger than 2GB.

agramfort, 2022-08-04:
You mean with just 

Pip install mne ?

I would say no. Just install mne with the hdf5 variant

cbrnr, 2022-08-04:
Yes, I mean `pip install mne`. Right now, this means that users cannot load EEGLAB files at all.

So I take it that you prefer option (2) then? Or do you think it's OK how it is and we shouldn't change anything?

agramfort, 2022-08-04:
Eeglab is not part of the 80/20 trade off for me.

Just use:

pip install mne[hdf5]

For packages who need hdf5 support

cbrnr, 2022-08-04:
OK, but what about option (2) then? We could enable loading EEGLAB files without any additional dependency. I'd assume that >80% of files are non-HDF5. Currently, we cannot load these files.

agramfort, 2022-08-04:
For point 2 then if we can have a reader just using scipy that’s great but I suspect it’s a bit of work no? Ideally this effort should done in one place for the python community.

cbrnr, 2022-08-04:
We can use the existing `scipy.io.loadmat()` reader. I've already implemented the changes, it's not much, the only thing that needs some work is our tests. We'd have to test HDF5 and non-HDF5 EEGLAB files separately.

PR #11006 (2022-08-05): Use `scipy.io.loadmat()` fallback for EEGLAB reader
Fixes #11005. Tests need to be adapted to test HDF5 and non-HDF5 files separately.
closed 2022-08-05 by larsoner