mne-tools/mne-python #11013: EEGLAB epoch export, latency error

arnodelorme, 2022-08-07:
Using the attached epoch file, trying to export to EEGLAB leads to this error

```python
epochs_ar2['stimulus/oddball_with_reponse'].
epochs_ar2['stimulus/oddball_with_reponse'].export('test.set', fmt='eeglab')
/Users/arno/miniconda3/envs/p38env/lib/python3.8/site-packages/eeglabio/epochs.py:102: UserWarning: Invalid event latencies, ignoring for export.
  warnings.warn("Invalid event latencies, ignoring for export.")
```

Also EEGLAB data is exported as double precision while the standard
 EEGLAB format (and MNE fif format) is simple precision. Would save half the space.

[test-epo.fif.zip](https://github.com/mne-tools/mne-python/files/9277398/test-epo.fif.zip)

cbrnr, 2022-08-07:
Hi Arno! We have just added support for this in #11000, did you already try this with the latest dev version and `eeglabio` 0.0.2?

arnodelorme, 2022-08-07:
Ah OK, thanks. I tried the MNE version you can get with pip install.

cbrnr, 2022-08-08:
I just tried with the latest main, but the problem still exists (I get the warning). In addition, I cannot read the exported file because of the following error:

```
In [11]: test = mne.io.read_epochs_eeglab("/Users/clemens/test.set")
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Input In [11], in <cell line: 1>()
----> 1 test = mne.io.read_epochs_eeglab("/Users/clemens/test.set")

File ~/Projects/mne-python/mne/io/eeglab/eeglab.py:317, in read_epochs_eeglab(input_fname, events, event_id, eog, verbose, uint16_codec)
    265 @fill_doc
    266 def read_epochs_eeglab(input_fname, events=None, event_id=None,
    267                        eog=(), verbose=None, uint16_codec=None):
    268     r"""Reader function for EEGLAB epochs files.
    269 
    270     Parameters
   (...)
    315     .. versionadded:: 0.11.0
    316     """
--> 317     epochs = EpochsEEGLAB(input_fname=input_fname, events=events, eog=eog,
    318                           event_id=event_id, verbose=verbose,
    319                           uint16_codec=uint16_codec)
    320     return epochs

File <decorator-gen-278>:12, in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, eog, verbose, uint16_codec)

File ~/Projects/mne-python/mne/io/eeglab/eeglab.py:519, in EpochsEEGLAB.__init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, eog, verbose, uint16_codec)
    517 event_name.append(event_type)
    518 # store latency of only first event
--> 519 event_latencies.append(events[ev_idx].latency)
    520 ev_idx += len(ep.eventtype)
    521 warn_multiple_events = True

IndexError: list index out of range
```

@larsoner @jackz314 I assume this is because the changes were not ready to be merged? https://github.com/mne-tools/mne-python/pull/11000#issuecomment-1204905141

@arnodelorme I will open a feature request to add a new parameter `precision` (which can be `"single"` or `"double"`) to our export function.

jackz314, 2022-08-08:
@arnodelorme So the warning occurs because the events have invalid onset latencies that exceed the total length of the data. 

In your example, I checked that the max latency in `events` is 193208, while the total length of the data is only 28527 samples. I'm not sure why it's off by that much, but this means the latencies cannot be exported because they're invalid.

@cbrnr the error is caused by a bug in the export function when annotation onsets are invalid (in this case onsets at ~754 seconds when the data is only ~111 seconds long). I will add a check that discards these invalid annotations, although I'm curious why this would happen in the first place.

arnodelorme, 2022-08-08:
Interesting. I will look up why the latency is off for this dataset. Thank you.

jackz314, 2022-08-08:
@cbrnr I added checks for these invalid onsets to main and tested it, it should work now.

cbrnr, 2022-08-09:
Thanks @jackz314! @arnodelorme please report back if you find out what the problem was. If the latencies exceed the signal duration, issuing a warning is the expected behavior.

I'm a bit concerned that exporting still works, but then importing fails. Has this been fixed with your changes @jackz314?

jackz314, 2022-08-09:
Importing should work as well now in the main branch, since I fixed the epoch structure exported along with a couple other things for when there are invalid onsets.

larsoner, 2022-10-21:
Sounds like this has been fixed so I'll close. Feel free to reopen if I'm mistaken

closed 2022-10-21 by larsoner